thesofproject/sof #1283: [CML]Paused aplay can't be resumed normally after DUT wake up from S3

ClarexZhou, 2019-04-19:
**Describe the bug**
Aplay resume failed once S3 and wake up from pause status.
2/5:  Failed for "write error:Input/output errror" 
3/5:  Resumed process bar display playing, but no audio output.

**To Reproduce**
1.  aplay -Dhw:0,3 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i
2. press SPACE bar to pause
3. S3 and wake up via "sudo rtcwake -m mem -s 10"
4. press SPACE bar to resume aplay, check

**Expected behavior**
Aplay should be resumed successfully and audio output normally.

**Environment**
Kernel Repo:
https://github.com/RanderWang/linux/commits/cml-validation-20190419
Made from:
RanderWang/linux@2de29fa

RanderWang/linux@d04ccf4
RanderWang/linux@bfb0729
RanderWang/linux@8a2b13a
RanderWang/linux@63ceb8f
RanderWang/linux@db89510
RanderWang/linux@066e6f4
RanderWang/linux@bfb3fcb
RanderWang/linux@edc012c
RanderWang/linux@a9f0431
RanderWang/linux@d5113b0
RanderWang/linux@aeccfd5
RanderWang/linux@22999bd
RanderWang/linux@c827ffb
Sof Repo: https://github.com/bardliao/sof/tree/whl-daliy-20190419
Made from: bardliao@ 4c6f6f7

bardliao/sof@90a6efd
bardliao/sof@5ef0713
bardliao/sof@50bd96d
bardliao/sof@fdf1eda
bardliao/sof@f135dd9
bardliao/sof@e8d5622

dmesg:
[ 7213.411021] sof-audio-pci 0000:00:1f.3: pcm: open stream 3 dir 0
[ 7213.411025] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[ 7213.411027] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[ 7213.411030] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[ 7213.411534] sof-audio-pci 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
[ 7213.411544] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 3 dir 0
[ 7213.411550] sof-audio-pci 0000:00:1f.3: generating page table for 000000002d38bc1d size 0xffc0 pages 16
[ 7213.411572] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[ 7213.411574] sof-audio-pci 0000:00:1f.3: periods:4
[ 7213.411590] sof-audio-pci 0000:00:1f.3: stream_tag 1
[ 7213.411602] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 7213.411771] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 7213.411779] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[ 7213.413138] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 1
[ 7213.413148] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 7213.413300] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 7213.413306] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[ 7213.905078] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 7213.905104] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 7214.404993] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 7214.405021] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 7216.564166] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 3
[ 7216.564181] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 7216.564287] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 7216.564295] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=3
[ 7217.404911] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[... 653 lines cut ...]
[ 7247.992674] sof-audio-pci 0000:00:1f.3: periods:4
[ 7247.992763] sof-audio-pci 0000:00:1f.3: stream_tag 1
[ 7247.992776] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 7247.992935] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 7247.992944] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[ 7247.999132] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[ 7247.999144] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 7247.999288] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 7247.999294] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[ 7248.458067] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 7248.458094] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 7248.958026] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 7248.958054] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 7263.201559] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[ 7263.201575] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 7263.201699] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 7263.201710] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[ 7263.201776] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[ 7263.201780] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 7263.201900] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 7263.202010] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[ 7263.957973] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 7263.957997] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[dmesg.log](https://github.com/thesofproject/sof/files/3097494/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3097495/logger.log)

ClarexZhou, 2019-04-19:
Same issue on arecord

lgirdwood, 2019-04-19:
@ClarexZhou this looks like a driver issue. I assume this works when the stream is not paused before suspend ? Can you also check this on other targets. I can see TRIGGER_START which I would not expect to see if the stream is paused. @plbossart @ranj063 fyi,

ClarexZhou, 2019-04-22:
It works OK when the stream is not paused before suspend.
Also:This issue cannot be reproduced on APL UP2 PCM512x platform.
Test Recipe
date: 2019-04-22
sof master: 0c93261
kernel sof-dev: 90860434
tplg: sof-apl-pcm512x.tplg

lgirdwood, 2019-04-23:
@mengdonglin fyi, this is a driver issue and it's unassigned

keqiaozhang, 2019-04-24:
This issue also can be reproduced on WHL with HDMI playback, but cannot be reproduced with I2S playback(headset playback).
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1         PIPE 4.21      9478057.656250   9478058.000000 src/audio/pipeline.c:789    pipeline_xrun_recover()
    0      1       VOLUME           9479258.177083      1200.520874   src/audio/volume.c:474    volume_copy() error: sink component buffer has not enoug   h free bytes for copy
    0      1         PIPE           9479302.500000        44.322918 src/audio/pipeline.c:681    pipeline_copy() error: ret = -5, start->comp.id = 20, di   r = 1
```
[dmesg.log](https://github.com/thesofproject/sof/files/3110731/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3110732/logger.log)
[logger-t.log](https://github.com/thesofproject/sof/files/3110733/logger-t.log)

Test recipe:
kernel:  sof-dev/2e94569
sof/tplg: master/e4c1651
tplg: sof-whl-rt5682.tplg

libinyang, 2019-04-24:
Bug has been root caused. It's a driver issue. I will submit a patch later.

lgirdwood, 2019-04-24:
@libinyang great, please create driver issues and link to this. I will close this now.

closed 2019-04-24 by wenqingfu

libinyang, 2019-04-25:
This is a driver issue, a new issue is filed in linux:
https://github.com/thesofproject/linux/issues/866

close this issue now.