thesofproject/sof #1296: [CML]DMIC16 cannot record 24/32 bit audio

ClarexZhou, 2019-04-23:
**Describe the bug**
DMIC16 display support 24/32 bit, but can't record, for "set_params:1363: Unable to install hw params"

**To Reproduce**
arecord -Dhw:0,7 -r16000 -c2 -fs24_le a.wav -vv -i

**Expected behavior**
Should record successfully.

**Environment**
Kernel repo	https://github.com/thesofproject/linux/tree/release/sof-v5.0
Made from:
9086043 [thesofproject/linux: topic/sof-dev]
    +575038e
    +	53cfc0d
    +	39d0823
    +	1b65a44
    +	98554b9
    +	1a44fac
    +	6e4d133
    +	2a816f7
    +	569f460
    +	f273121
    +	8cbbf45
    +	8de4c6
    +	f891831
    +	e5ced5b
    +	0ad31d4
    +	f5f69bd
    +	5df2c93
    +	8594d32
    +	c48713b
	
Firmware repo	https://github.com/RanderWang/sof/tree/cml-linux-20190423
	Made from:
	0c932613  [thesofproject/sof: master]
    +	adf4db3 [RanderWang/sof: cml-linux-20190423]
    +	1c8cbd1 [RanderWang/sof: cml-linux-20190423]
	
Topology: 	https://github.com/RanderWang/sof/tree/cml-linux-20190423
	file:  tools/topology/sof-hda-generic.m4,  sof-hda-generic-2ch.tplg

```
test@test-Inspiron-5490:~$ arecord -Dhw:0,7 /dev/null --dump-hw-params
Recording WAVE '/dev/null' : Unsigned 8 bit, Rate 8000 Hz, Mono
HW Params of device "hw:0,7":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: 16000
PERIOD_TIME: [1500 255000]
PERIOD_SIZE: [24 4080]
PERIOD_BYTES: [192 16320]
PERIODS: [2 16]
BUFFER_TIME: [3000 1023000]
BUFFER_SIZE: [48 16368]
BUFFER_BYTES: [192 65472]
TICK_TIME: ALL
--------------------
arecord: set_params:1299: Sample format non available
Available formats:
- S16_LE
- S24_LE
- S32_LE
test@test-Inspiron-5490:~$ arecord -Dhw:0,7 -r16000 -c2 -fs24_le a.wav -vv -i
Recording WAVE 'a.wav' : Signed 24 bit Little Endian, Rate 16000 Hz, Stereo
arecord: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S24_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 16000
PERIOD_TIME: 124500
PERIOD_SIZE: 1992
PERIOD_BYTES: 15936
PERIODS: (4 5)
BUFFER_TIME: 499500
BUFFER_SIZE: 7992
BUFFER_BYTES: 63936
TICK_TIME: 0
```
[log.zip](https://github.com/thesofproject/sof/files/3106755/log.zip)

emilchudzik, 2019-04-23:
Why did you use 24/32bit format on DMIC16 (16bit)?
Without alsa conversion you will be hw params errors, this is expected for your command. 
For me not a bug.

closed 2019-04-23 by emilchudzik

ClarexZhou, 2019-04-24:
Bug is reported for Display support 24/32 bit format when checking via "arecord -Dhw:0,7 /dev/null --dump-hw-params"
Available formats:
- S16_LE
- S24_LE
- S32_LE
Also DMIC32 can use all 16,24 and 32 bit format normally.