thesofproject/sof #1297: [CML]Volume changed response delay about 3-5s during DMIC recording

ClarexZhou, 2019-04-23:
**Describe the bug**
Volume changed response delay about 3-5s during DMIC recording

**To Reproduce**
1. arecord -Dhw:0,6 -r48000 -c2 -fs16_le a.wav -vv -i
2. Open another Terminal, alsamixer->F4, quickly change volume between 0-max100 (PGA10.0 for Dhw:0,6 )
3. check arecord process (recorded db *%)  during volume change

**Expected behavior**
Volume change should become effective quickly

**Environment**
Kernel repo	https://github.com/thesofproject/linux/tree/release/sof-v5.0
Made from:
9086043 [thesofproject/linux: topic/sof-dev]
+575038e
+	53cfc0d
+	39d0823
+	1b65a44
+	98554b9
+	1a44fac
+	6e4d133
+	2a816f7
+	569f460
+	f273121
+	8cbbf45
+	8de4c6
+	f891831
+	e5ced5b
+	0ad31d4
+	f5f69bd
+	5df2c93
+	8594d32
+	c48713b

Firmware repo	https://github.com/RanderWang/sof/tree/cml-linux-20190423
Made from:
0c93261 [thesofproject/sof: master]
+	adf4db3 [RanderWang/sof: cml-linux-20190423]
+	1c8cbd1 [RanderWang/sof: cml-linux-20190423]

Topology: https://github.com/RanderWang/sof/tree/cml-linux-20190423
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

[volumeChange_Dmesg.log](https://github.com/thesofproject/sof/files/3106934/volumeChange_Dmesg.log)
[volumerChange_logger.log](https://github.com/thesofproject/sof/files/3106935/volumerChange_logger.log)

singalsu, 2019-04-23:
This is because the firmware volume uses a linear ramp that is hardcoded to be 250 ms from mute to 0 dB. Now that capture max is +30 dB (~32 times unity gain - 0 dB) the ramp could take up to 8 seconds.

I did not include modification to volume scale increase PR because this is a separate issue. To fix this there are a number of options

- Implement to driver and topology and FW the volume ramp customization, the tokens exist in ASoC. The fix could use log ramp type that's by nature faster or faster linear ramp parameters for high gain volume instances.

- In FW as small update move ramp time compilation to vol_work() launch code to on the fly compute the step size to maintain 250 ms ramp

@lgirdwood @tlauda @plbossart What would you prefer?

lgirdwood, 2019-04-23:
@singalsu IPC supports ramp meta data 
```c
/* volume ramping types */
enum sof_volume_ramp {
	SOF_VOLUME_LINEAR	= 0,
	SOF_VOLUME_LOG,
	SOF_VOLUME_LINEAR_ZC,
	SOF_VOLUME_LOG_ZC,
};

/* generic volume component */
struct sof_ipc_comp_volume {
	struct sof_ipc_comp comp;
	struct sof_ipc_comp_config config;
	uint32_t channels;
	uint32_t min_value;
	uint32_t max_value;
	uint32_t ramp;		/**< SOF_VOLUME_ */
	uint32_t initial_ramp;	/**< ramp space in ms */
} __attribute__((packed));
```
So we are probably missing from topology, kernel parsing and FW volume init ?

singalsu, 2019-04-23:
@lgirdwood Yes, those are missing.

emilchudzik, 2019-04-23:
In our python tests (windows + slim driver) we use ramp meta data and it works good.

Below example for test with changing volume on DMIC:

```
  min_value=vol.VOL_MIN,
  max_value=vol.VOL_MAX,
  ramp=VolumeRamp.SOF_VOLUME_LINEAR,
  initial_ramp=1
...
  vol_lvls = [vol.VOL_MAX, vol.VOL_MAX//4, vol.VOL_MAX//2, int(vol.VOL_MAX*0.75)]
```

[image]

For SSP it works also fine.

singalsu, 2019-04-23:
@emilchudzik What is your VOL_MAX? If it's 65536 (linear 1.0 gain) the ramp time is the 250 ms as earlier. The slow settling of gain happens when volume is set much higher.

zrombel, 2019-04-23:
For Python tests we are using VOL_MAX = 65536, and in test that waveform comes from volume sequence is: 65536, 16384, 32768, 49152.

zrombel, 2019-04-23:
I've set VOL_MAX to 524288 and volume sequence to [524288, 131072, 262144, 393216] and now ramp has length of about 1,5s. 
[image]

singalsu, 2019-04-23:
Max volume for Q8.16 is 2^23 - 1 = 8388607 so the ramp can take longer. However that's a very large gain (+42 dB or 128x linear) and practical gains would be always lower and ramp times a bit shorter. Though this is issue is relevant. If a mic would need a high gain e.g. start of captured utterance would be very silent with such slow ramp.

singalsu, 2019-04-24:
@kv2019i FYI

keyonjie, 2019-04-24:
@singalsu maybe we should limit the maximum ramp time to e.g. 250ms, no matter how max volume we target to tune to? It introduce bad user experience that you want to change a volume but you have to wait for 3 seconds until it is done.

singalsu, 2019-04-24:
@keyonjie Yes that would be a simple and easy to do fix just in FW side. I'd like to get feedback what approach we should take to address this.

kv2019i, 2019-04-25:
@lgirdwood wrote:
> 	uint32_t ramp;		/**< SOF_VOLUME_ */
> 	uint32_t initial_ramp;	/**< ramp space in ms */

It would seem we are not implementing the 'initial_ramp' correctly. Or if I understant that correctly -- the comment indicates this is the ramp length in milliseconds. Not sure why the word "initial" is used in the field name.

If I understand that right, the tests @zrombel  has done are not quite correct as you are specifying an initial ramp of 1ms, but you actually get a 250+ms ramp and the test still passes (it should not).

keqiaozhang, 2019-04-25:
This issue also can be reproduced on GLK, I changed the DMIC to volume pipeline in M4 file. and volume change takes a few seconds to be effective, It's a slow gradual process，please refer to the following screenshot.
[image]

steps:
1. set the DMIC vloume kctl PGA to 20 and play the sine wav on speaker.
   $ amixer -c0 cset name='PGA4.0 4 Master Capture Volume' 20
   $ aplay -Dhw:0,0 -f dat -c 2 s16_le_sine_60s.wav
2. After a few seconds, open another terminal, do DMIC recording and change the volume to 80 in parallel.
    $ arecord -Dhw:0,99 -f s16_le -c 4 -r 48000 dmic-test.wav -d 25 &  amixer -c0 cset name='PGA4.0 4 Master Capture Volume' 80
3. The picture shows that it took more than 6 seconds to reach the top.

Test recipe:
kernel: sof-dev/f7e9d4
sof: : https://github.com/RanderWang/sof/tree/cml-linux-20190425
tplg: sof-glk-da7219.tplg (modified the DMIC to the volume pipeline)

lgirdwood, 2019-04-25:
@kv2019i not sure why we have ```initial_``` prefix either. This value is intended to be the gap between volume ramp increments in ms.

singalsu, 2019-04-25:
OK, I can start work with topology, kernel, FW patches set to implement the ramp IPC (just linear ramp type, other types later). If we would need a quick workaround fix in FW it can be done with quite low amount of work.

zrombel, 2019-04-25:
We have tested how change of parameters **uint32_t ramp** and **uint32_t initial_ramp** affects the waveform, but from what we can observe it doesn't change anything - we tried all SOF_VOLUME_ defines and a wide range of initial_ramp <1 : 10000>. I looked in to FW code and it seems that those parameters aren't used, so this functionalities may be not implemented yet. Or maybe those parameters are used indirectly and that just me couldn't find them in code?

singalsu, 2019-04-25:
@zrombel It's not implemented. Volume ramp is only defined by macros in volume.h.

zrombel, 2019-04-25:
@kv2019i tests were made to check if volume scale the input signal properly in static state. When ramp setting will be implemented we will prepare test to cover that functionality.

kv2019i, 2019-04-25:
@zrombel wrote:
> @kv2019i tests were made to check if volume scale the input signal properly in static state. When ramp setting will be implemented we will prepare test to cover that functionality.

Ack, sounds good, thanks Seppo and Zbigniew!

wenqingfu, 2019-04-27:
so is the conclusion "not a bug"? or what's the plan next? @singalsu @lgirdwood

singalsu, 2019-04-29:
@wenqingfu I'll implement PR set to support volume ramp tokens passing to FW volume component. Then we can specify the ramp per volume instance and have similar fast settling for high gain volume instances as we have for current attenuate-only volume instances.

singalsu, 2019-04-30:
Small update, I will have one day break due to local bank holiday. The driver and topology defines the tokens so I was able to compile a new topology with volume ramp specification. However it looks like in kernel the ramp tokens are not passed to FW so I'll debug the driver next (the FW side changes are known and quite simple).

singalsu, 2019-05-03:
@lgirdwood @tlauda @wenqingfu I get now the volume ramp parameters passed to DSP. The kernel is as such OK. I just had problems with my m4 macros edits in topology to attach the ramp tokens for volume controllers.

Next I will work with FW part.

lgirdwood, 2019-05-03:
@singalsu tahnsk, any PRs for topology ?

singalsu, 2019-05-03:
@lgirdwood Yep there will be topology PR. I seem to need to update nearly all topologies due to parameter add for pga.m4. I've updated so far only one topology for testing.

I'm still wondering if I should provide from driver the volume instance specific vol max/min values. Currently I'm working with code that makes every transition ramp take the topology defined time (250 ms). That's a bit slow for very small volume change perhaps.

With max/min information I could make the worst case transition to take the defined time and a smaller transition a fraction of it (constant rate ramp). However retrieving that information within driver can be a bit tricky. I'm not sure about order of things in parsing topology.

lgirdwood, 2019-05-03:
@singalsu topology should define ramp for each PGA control. This should be loaded by driver during topology loading and sent to FW. Parsing order doesn't matter as volume is all parsed at once and wont be sent to FW unless all PGA are completely parsed.

ranj063, 2019-05-03:
@singalsu volume ramp step and type are already included and parsed in the driver. So you could just add the tokens in tpology and everything should be taken care of, right?

singalsu, 2019-05-03:
@lgirdwood OK, then I should be able to get the linear max/min values from TLV volume scale computation.

singalsu, 2019-05-03:
@ranj063 Yes, I have that part working already. I messed a day or two with m4 macros but I got it finally. Now the tokens appear correctly to firmware.

wenqingfu, 2019-05-06:
any PR submitted yet? @singalsu

singalsu, 2019-05-06:
I have now a working solution, next cleanup of code and doing the update to rest of topologies (I've updated and tested only sof-apl-dmic-2ch.tplg). I gave up with logarithmic ramp FW code so it will be only linear ramp with topology adjustable ramp length.

Logarithmic ramp and and the other ramp types implementation could be added as FW enhancement issues into Github. The logaritmic update is simple but length estimation is somewhat tricky.

singalsu, 2019-05-06:
I made an early [RFC] PR #1373 to review the volume enhance and start testing.

YvonneYang2, 2019-05-10:
Test on CML labtop, no obvious volume delay via DMIC capture.

Test enviroment as below:
Rander UCM https://github.com/RanderWang/intel-UCM  
Kernel repo：d698c4f
https://github.com/thesofproject/linux/tree/release/sof-v5.0
Firmware repo: 030c81c
https://github.com/thesofproject/sof/commits/master |  
Topology: https://github.com/thesofproject/sof/commits/master . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg