thesofproject/sof #1298: [BUG] SOF_FW_MAX_SIZE - do we really need such defines?

mrajwa, 2019-04-23:
**Describe the bug**
If you try to link external library with our firmware you will get an error message saying

> sof section `.data' will not fit in region `sof_fw'

So your first attempt will be to increase the SOF_FW_MAX_SIZE define in memory.h. Doing so will cause some other defines (like i.e HEAP_BUFFER_COUNT) to go negative!

**To Reproduce**
Increase SOF_FW_MAX_SIZE by 0x10000 and try to compile firmware.

**Expected behavior**
No defines limiting firmware size.
 
**Impact**
Slows down integration of third party components.

**Environment**
- Master branch
- Topology doesn't matter 
- Apollo lake platform

[image]

Why do we keep such defines in platform? We should let linker handle memory boundaries calculation.

lgirdwood, 2019-04-23:
@mrajwa these were needed for BYT -> BDW since the RAM was segregated into IRAM and DRAM.  Not needed on SKL onwards though.

abonislawski, 2019-05-06:
@mrajwa  not a bug, just a limit of memory.

The same will happen if you will just add 0x1000000 but this doesnt mean there is a bug, there is just no memory for that. You can increase memory banks quantity in kconfig (just to check and build but this is already a max value for apl).

SOF_FW_MAX_SIZE is needed because we need to specify a size of memory segment to linker script. Previously there were 3 max sizes for text/rodata/bss.

In your case HEAP_BUFFER_COUNT go to negative because HEAP_BUFFER_SIZE is the first negative (wrong) size. Thats because HEAP_BUFFER_BASE is using SOF_FW_MAX_SIZE (so BASE is placed over memory limit) and SOF_STACK_END which is at the end of memory (no connection to SOF_FW_MAX_SIZE).
So HEAP_BUFFER_SIZE = (SOF_STACK_END - HEAP_BUFFER_BASE) will result in a negative value.

The only question is that if we should add a clear check for that? @mmaka1 @jajanusz 
(but not sure where)

jajanusz, 2019-05-07:
@abonislawski We can add some static asserts but idk if it is ok regarding our kernel-like conventions. I saw there are like 3 `STATIC_ASSERT`s in kernel code but idk why.

jajanusz, 2019-06-21:
`SOF_FW_MAX_SIZE` is just limited by HP memory size. There is not point in increasing it manually if it will not fit into memory. If your platform has more memory/banks it will try to use as much as it can.

closed 2019-06-21 by jajanusz