thesofproject/sof #1303: [CML]Laptop speaker always no audio output after plug in headset.

ClarexZhou, 2019-04-24:
**Describe the bug**
Laptop speaker always no audio output after plug in headset, even after unmute speaker and increase speaker volume manually via alsamixer. Only audio output on headset.

**To Reproduce**
1. 3mm Headset is pluged in
2 .aplay -Dhw:0,0 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i
3. Open another Terminal, 'alsamixer' to unmute Speaker and increase Speaker volume manually, check

**Expected behavior**
Audio output should via both Laptop speaker and Headset at same time

More info:
Error seen not matter pulseaudio is off or on.

Environment
Kernel repo	https://github.com/thesofproject/linux/tree/release/sof-v5.0
Made from:
9086043 [thesofproject/linux: topic/sof-dev]
+575038e

53cfc0d
39d0823
1b65a44
98554b9
1a44fac
6e4d133
2a816f7
569f460
f273121
8cbbf45
8de4c6
f891831
e5ced5b
0ad31d4
f5f69bd
5df2c93
8594d32
c48713b
Firmware repo	https://github.com/RanderWang/sof/tree/cml-linux-20190423
Made from:
0c93261 [thesofproject/sof: master]

adf4db3 [RanderWang/sof: cml-linux-20190423]
1c8cbd1 [RanderWang/sof: cml-linux-20190423]
Topology: https://github.com/RanderWang/sof/tree/cml-linux-20190423
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

Log:
```
[  317.803966] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  317.803988] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  317.804208] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  317.804218] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  318.741540] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  318.741570] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  322.086783] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.087099] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.087120] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.087457] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.087479] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.087711] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.087737] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.087998] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088019] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088173] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088188] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088381] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088397] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088628] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088649] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088915] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.088940] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.089117] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: SET_VALUE: GET_VALUE
[  322.741619] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  322.741646] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  347.893586] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  347.893607] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  347.893757] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  347.893768] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[  347.893947] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  347.893957] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  347.894162] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  347.894413] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
```
[SpeakerNoOutputAfterPlugInHeadset_dmesg.log](https://github.com/thesofproject/sof/files/3110190/SpeakerNoOutputAfterPlugInHeadset_dmesg.log)
[SpeakerNoOutputAfterPlugInHeadset_logger.log](https://github.com/thesofproject/sof/files/3110191/SpeakerNoOutputAfterPlugInHeadset_logger.log)

ClarexZhou, 2019-04-24:
@keqiaozhang @stevyan Could you please help to check on other platforms (GLK, ICL, UP2, BYT)? Thanks.

keqiaozhang, 2019-04-24:
@ClarexZhou 
This issue can not be reproduced on GLK. Other platforms don't have speaker.

ranj063, 2019-04-24:
@keqiaozhang seems to suggest a UCM problem maybe?

libinyang, 2019-04-24:
Root caused. We need disable “Auto-Mute Mode” in alsamixer.

mengdonglin, 2019-04-24:
@libinyang How to" disable “Auto-Mute Mode” in alsamixer"?
@RanderWang Need to cover this in UCM?

libinyang, 2019-04-24:
Just select the auto mute item and press arrow up or arrow down keys to select.

emilchudzik, 2019-04-24:
This is expected behaviour, even on my private laptop or corporate laptop, or even on phones with android -> When you connect headphone by jack or pair by bluetooth then speaker is muted. This is default.

Where did the idea for this scenario test come from?

ClarexZhou, 2019-04-24:
Issue raised for different behavior with previous builds.
Root cause for System default setting.   Close.

closed 2019-04-24 by ClarexZhou