thesofproject/sof #1305: [CML]HDA Analog : Recorded audio has serious distortion and pop noise when volume to max.

ClarexZhou, 2019-04-24:
**Describe the bug**
Recorded audio has serious distortion and pop noise when volume to max.  
Much better when change headset Mic boost to 0db.

**To Reproduce**
1. Change Capture, Headset Mic Boost and HDA Analog volume(PGA 2.0 2 Master) to max
2. Record  via ''arecord -Dhw:0,0 -r48000 -c2 -fs16_le maxvolume.wav'' and play this audio file to check quality 
3. Change Headset Mic Boost to 0db and others keeps Max and record again to check.

**Expected behavior**
Recorded audio should be clear.

**Environment**

Kernel repo | https://github.com/RanderWang/linux/tree/cml-validation-20190424
  Made from:
  2e94569 [thesofproject/linux: topic/sof-dev]

85a86d9 [RanderWang/linux: cml-validation-20190424]
b51284b[RanderWang/linux: cml-validation-20190424]
8f41f15 [RanderWang/linux: cml-validation-20190424]
202216f [RanderWang/linux: cml-validation-20190424]
ce16096 [RanderWang/linux: cml-validation-20190424]
4459c97 [RanderWang/linux: cml-validation-20190424]
4f4bfb1 [RanderWang/linux: cml-validation-20190424]
0bc8e92 [RanderWang/linux: cml-validation-20190424]
4de6c77 [RanderWang/linux: cml-validation-20190424]
7ec78da [RanderWang/linux: cml-validation-20190424]
30d5691 [RanderWang/linux: cml-validation-20190424]
98d6916 [RanderWang/linux: cml-validation-20190424]
 
Firmware repo | https://github.com/RanderWang/linux/tree/cml-validation-20190424
  Made from:
  0c93261 [thesofproject/sof: master]
e3ade00 [RanderWang/sof: cml-linux-20190424]
08d40fc [RanderWang/sof: cml-linux-20190424]

[image]
[maxVolume_Recorded Video.zip](https://github.com/thesofproject/sof/files/3111376/maxVolume_Recorded.Video.zip)

lgirdwood, 2019-04-24:
@ClarexZhou is recorder audio good quality at lower gain ? What about when Mic Boost is disabled ?
I suspect Mic Boost is not needed on this device.

mengdonglin, 2019-04-24:
@lgirdwood @singalsu Canonical thinks sound of DMIC capture is weak without Mic Boost.
Does Mic Boost usually support some kind of Auto Gain Control ?

lgirdwood, 2019-04-24:
@mengdonglin check with DMIC/codec vendor. Maybe @bardliao will know.

ClarexZhou, 2019-04-25:
Audio quality is good at lower gain. eg. Capture volume 58% 6db, Headset Mic boost 53% 20db and PG2.0 2 at 40%. 
And still has quality issue after change Mic boost is 0, and keed PGA2.0 2 and Capture volume at Max.

keqiaozhang, 2019-04-25:
This issue also can be reproduced on GLK if all related volumes are set to maximum。
Test recipe:
sof/tplg: master/395a34
kernel: sof-dev/f7e9d4a

lgirdwood, 2019-04-25:
@ClarexZhou @keqiaozhang please be careful with adding gain above 0dB to any Mic signal as this can introduce signal clipping as gain in the pipeline is cumulative. i.e. any cumulative pipeline gain above 0dB may clip depending on input signal size. So if Mic boots at 20dB is needed then we will probably need to keep the other capture volumes at 0dB or less. The intention is that the user can configure capture volume to get best microphone performance.

closed 2019-04-25 by lgirdwood

singalsu, 2019-04-25:
When there is analog volume control available in codec for microphone use it to set desired gain and keep digital gain for capture as 0 dB. The signal level in SOF digital parts should be high (but not clip) and attenuation/amplification done near to endpoint. 

The digital gain in capture path is useful when there is no codec gain available like in PCH or SoC direct attach DMIC case.

It would be useful to have different topologies for different type of capture endpoints. With codec gain available the volume scale of capture digital gain could be max 0 dB (we could has two volume capture pipes, one with max +30 dB and one with max +0 dB) instead of current +30 dB. Or the volume could be even omitted from topology (use pass-through capture pipe).

Upper level topologies should be specific to platform + used audio peripheral chips to avoid risk for user to set gains in harmful way.