thesofproject/sof #1307: [BUG][CML]HDMI aplay still work for about 5mins after plugout HDMI and restart aplay

ClarexZhou, 2019-04-24:
**Describe the bug**
HDMI aplay still work for about 5mins after plugout HDMI and restart aplay

**To Reproduce**
1.  Connect HDMI
2.  Play music via HDMI: aplay -Dhw:0,3 -r48000 -c2 -fs16_le a.wav -vv -i 
3.  Disconnect HDMI
4.  Stop previous aplay and start new aplay, check

**Actual behavior**
aplay to HDMI still works after disconnected

**Expected behavior**
Should detect HDMI is disconnected.

**Environment**
Kernel repo | https://github.com/RanderWang/linux/tree/cml-validation-20190424
  Made from:
  2e94569 [thesofproject/linux: topic/sof-dev]

85a86d9 [RanderWang/linux: cml-validation-20190424]
b51284b[RanderWang/linux: cml-validation-20190424]
8f41f15 [RanderWang/linux: cml-validation-20190424]
202216f [RanderWang/linux: cml-validation-20190424]
ce16096 [RanderWang/linux: cml-validation-20190424]
4459c97 [RanderWang/linux: cml-validation-20190424]
4f4bfb1 [RanderWang/linux: cml-validation-20190424]
0bc8e92 [RanderWang/linux: cml-validation-20190424]
4de6c77 [RanderWang/linux: cml-validation-20190424]
7ec78da [RanderWang/linux: cml-validation-20190424]
30d5691 [RanderWang/linux: cml-validation-20190424]
98d6916 [RanderWang/linux: cml-validation-20190424]
 
Firmware repo | https://github.com/RanderWang/linux/tree/cml-validation-20190424
  Made from:
  0c93261 [thesofproject/sof: master]
e3ade00 [RanderWang/sof: cml-linux-20190424]
08d40fc [RanderWang/sof: cml-linux-20190424]

No obvious issue on log.
```
[HDMIstillWorkAfterDisconnlogger.log](https://github.com/thesofproject/sof/files/3111718/HDMIstillWorkAfterDisconnlogger.log)
[HDMIstillWorkAfterDisconnect_dmesg.log](https://github.com/thesofproject/sof/files/3111719/HDMIstillWorkAfterDisconnect_dmesg.log)

[  563.865753] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[  563.865925] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[  563.865937] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[  563.866108] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[  563.866118] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  563.866194] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  563.871861] sof-audio-pci 0000:00:1f.3: pcm: open stream 3 dir 0
[  563.871866] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  563.871869] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  563.871871] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  563.872464] sof-audio-pci 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
[  563.872479] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 3 dir 0
[  563.872485] sof-audio-pci 0000:00:1f.3: generating page table for 0000000074a42ddf size 0xffc0 pages 16
[  563.872524] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  563.872527] sof-audio-pci 0000:00:1f.3: periods:4
[  563.872551] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  563.872564] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  563.872841] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  563.872848] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[  563.874958] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 1
[  563.874973] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  563.875159] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  563.875170] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  564.364409] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  564.364438] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  564.864356] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  564.864384] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  570.946599] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 0
[  570.946614] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  570.946810] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  570.946823] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[  570.946971] sof-audio-pci 0000:00:1f.3: pcm: free stream 3 dir 0
[  570.946980] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  570.947083] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  570.947186] sof-audio-pci 0000:00:1f.3: pcm: close stream 3 dir 0
[  571.864278] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  571.864306] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  572.688707] sof-audio-pci 0000:00:1f.3: pcm: open stream 3 dir 0
[  572.688709] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  572.688711] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  572.688712] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  572.689071] sof-audio-pci 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
[  572.689080] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 3 dir 0
[  572.689083] sof-audio-pci 0000:00:1f.3: generating page table for 0000000074a42ddf size 0xffc0 pages 16
[  572.689107] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  572.689108] sof-audio-pci 0000:00:1f.3: periods:4
[  572.689122] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  572.689130] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  572.689323] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  572.689328] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[  572.690899] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 1
[  572.690906] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  572.691051] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  572.691054] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  573.364596] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  573.364626] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

lgirdwood, 2019-04-24:
@ClarexZhou this is a driver bug. FW has no way of knowing whether HDMI is connected. Please re-log as driver issue.

closed 2019-04-24 by wenqingfu

wenqingfu, 2019-04-25:
moved to thesofproject/linux#865