thesofproject/sof #1318: [BUG] Volume: Bit conversion from 32/32 to 24/32 glitch

wwittbrx, 2019-04-25:
### Description

FW commit 50bd96d introduces a glitch during bit conversion from 32/32 to 24/32 in slim driver tests. This happens over all samplerates and channel counts. Conversions from 32/32 to 16/16 work as expected.

The glitch looks like this (full file below):

[image]

The volume value is not set for this test (using default which is VOL_MAX from what I could gather). If we set volume to VOL_MAX - 1 before conversion, the recorded wave looks as expected.

### Reproduction

Run
test_14_ptp_bit_depth_conversion_loopback.py
TestPtpBitDepthConversionLoopback8000Hz32b32b1ch

### Recipe

**FW**
https://github.com/thesofproject/sof/commits/master
50bd96d

### Logs and files

[log_fail.zip](https://github.com/thesofproject/sof/files/3117477/log_fail.zip)
[log_success.zip](https://github.com/thesofproject/sof/files/3117478/log_success.zip)

singalsu, 2019-04-26:
Thanks @wwittbrx I'll fix this ASAP.

singalsu, 2019-04-26:
I've improved the cmocka volume test bench to test with larger numbers and I'm now able to repeat this error.

lgirdwood, 2019-04-26:
@singalsu what's the ETA for the fix ? If today I can pull into v1.3-rc3 ? otherwise we can get it into v1.3 final next week.

singalsu, 2019-04-26:
@lgirdwood The fix is in PR #1341.

wwittbrx, 2019-04-29:
I confirmed that PR #1341 fixes this issue, once it's merged and retested on master I will close this issue. Thanks.

wwittbrx, 2019-04-30:
Checked on master branch, commit 3262ac1. This is fixed now. Thank you, I will be closing this now.

closed 2019-04-30 by wwittbrx