thesofproject/sof #1329: [BUG][CML]Arecord via HDA Analog always shown busy, only system reboot can recover. (Only occurred one time)

ClarexZhou, 2019-04-26:
**Describe the bug**
Arecord via HDA Analog headset always shown busy, only system reboot can recover. (Only occurred one time)

**To Reproduce**
Boot up system, aplay via analog, stop and then arecord via analog headset, check

**Expected behavior**
arecord should work normally 

**Environment**
Kernel repo | https://github.com/RanderWang/linux/tree/cml-validation-20190426

 Made from:
  f7e9d4a [thesofproject/linux: topic/sof-dev]

f1ea666 [RanderWang/linux: cml-validation-20190426]
bebb024 [RanderWang/linux: cml-validation-20190426]
00ed6b5 [RanderWang/linux: cml-validation-20190426]
544ddb6 [RanderWang/linux: cml-validation-20190426]
d6d12c3 [RanderWang/linux: cml-validation-20190426]
5fd5c57 [RanderWang/linux: cml-validation-20190426]
cb621ca [RanderWang/linux: cml-validation-20190426]
77c1830 [RanderWang/linux: cml-validation-20190426]
cb33024 [RanderWang/linux: cml-validation-20190426]
6e0e8fb [RanderWang/linux: cml-validation-20190426]
   
Firmware repo | https://github.com/RanderWang/sof/tree/cml-linux-20190426
  Made from:
  76309da [thesofproject/sof: master]
dc3adeaf [RanderWang/sof: cml-linux-20190426]
53c7062 [RanderWang/sof: cml-linux-20190426]
   
Topology: | https://github.com/RanderWang/sof/tree/cml-linux-20190426
  file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

No error message on sof-logger.
```
[  674.932929] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[  674.932935] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  674.932938] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  674.932941] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  674.933175] sof-audio-pci 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
[  674.933186] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 0 dir 0
[  674.933193] sof-audio-pci 0000:00:1f.3: generating page table for 00000000cbf45e80 size 0xffc0 pages 16
[  674.933225] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  674.933227] sof-audio-pci 0000:00:1f.3: periods:4
[  674.933254] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  674.933272] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  674.933457] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  674.933465] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790604
[  674.939634] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  674.939647] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  674.939881] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  674.939892] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  675.559268] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  675.559297] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  680.463580] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  680.463600] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  680.463798] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  680.463814] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[  680.463894] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  680.463901] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  680.464024] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  680.464133] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  681.058814] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  681.058843] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  688.754508] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[  688.754514] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  688.754518] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  688.754521] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  688.754764] sof-audio-pci 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
[  688.754776] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 0 dir 0
[  688.754783] sof-audio-pci 0000:00:1f.3: generating page table for 00000000cbf45e80 size 0xffc0 pages 16
[  688.754815] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  688.754817] sof-audio-pci 0000:00:1f.3: periods:4
[  688.754838] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  688.754855] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  688.755019] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  688.755025] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790604
[  688.755398] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  688.755416] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  688.755579] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  688.755591] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  689.558923] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  689.558950] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  694.281084] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  694.281103] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  694.281240] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  694.281251] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[  694.281324] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  694.281330] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  694.281457] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  694.281588] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  695.059205] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  695.059235] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  704.098004] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[  704.098010] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  704.098014] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  704.098017] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  704.098240] sof-audio-pci 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
[  704.098251] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 0 dir 0
[  704.098258] sof-audio-pci 0000:00:1f.3: generating page table for 00000000cbf45e80 size 0xffc0 pages 16
[  704.098288] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  704.098290] sof-audio-pci 0000:00:1f.3: periods:4
[  704.098316] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  704.098334] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  704.098536] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  704.098543] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790604
[  704.098873] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  704.098891] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  704.099109] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  704.099122] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  705.059181] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  705.059208] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

[HDAAnalogArecordAlwasyBusy.log](https://github.com/thesofproject/sof/files/3120515/HDAAnalogArecordAlwasyBusy.log)
[logger.log](https://github.com/thesofproject/sof/files/3120516/logger.log)

lgirdwood, 2019-04-26:
@ClarexZhou looks like a driver bug. In general, anytime userspace has a file IO issue it's 90% likely to be a driver bug rather than a FW bug. Can you open this as driver issue and close this one.

ClarexZhou, 2019-04-28:
Close and track in https://github.com/thesofproject/linux/issues/881

closed 2019-04-28 by ClarexZhou