thesofproject/sof #1330: [BUG] CHT target is not linking after new features are added to a common code

lbetlej, 2019-04-26:
**Describe the bug**
The issue is reproducible with PR#949. The CHT code is not linking (BYT and all other targets link properly)

**To Reproduce**
Compile the CHT code with PR#949

**Expected behavior**
CHT/BYT code reviewed to remove unnecessary code or added slim feature configuration as a default one.

**Impact**
No impact for other platforms.

**Environment**
Compile the CHT code with PR#949
 Reproducibility Rate: always

**Screenshots or console output**
 https://sof-ci.01.org/sofpr/PR949/build1774/build/

lgirdwood, 2019-04-26:
Text size too small, lets increase it and deselect some Kconfig options.

lgirdwood, 2019-04-26:
@lbetlej see https://github.com/thesofproject/sof/pull/1338, will fix issue.

closed 2019-05-01 by lgirdwood