thesofproject/sof #1332: [RFC] ipc: _IPC_COPY_CMD macro usage

bkokoszx, 2019-04-26:
**Is your feature request related to a problem? Please describe.**
During handling IPC commands we utilize IPC_COPY_CMD macro for copying IPC in a safe way as it is written in code comments. What is the purpose of this solution? We declare new variable on a stack and after all we use pointer to global context variable ```_ipc```.

E.g.
In function: ```static int ipc_stream_pcm_params(uint32_t stream)```
there is a declaration: 
```struct sof_ipc_pcm_params pcm_params;
IPC_COPY_CMD(pcm_params, _ipc->comp_data);```
and after all we use pointer directly to _ipc global variable:
```err = pipeline_params(pcm_dev->cd->pipeline, pcm_dev->cd,(struct sof_ipc_pcm_params *)_ipc->comp_data);```

I'm aware that the assumption was to enable minor abi changes, but in our case if we have flexible arrays at the end of specific ipc structure (e.g. in  ```struct sof_ipc_ctrl_data```) we should always use pointer to a global context variable.

**Describe the solution you'd like**
Since we use pointer to global ```_ipc``` we do not need creating additional variables.

I would be grateful for explanations.

lgirdwood, 2019-04-26:
@bkokoszx see https://github.com/thesofproject/sof/pull/1336, we should be deprecating our globals to give more flexibility with allocation, positioning and use a sof context to store this stuff.

bkokoszx, 2019-05-06:
@lgirdwood 
Ok, thanks, I see. But still after those changes, should we use IPC_COPY_CMD macro like we used before e.g. ```IPC_COPY_CMD(pcm_params, sof->ipc->comp_data)```?
We can invoke functions like ```pipeline_params``` with ```sof->ipc->comp_data``` directly without copying context to local variables.

lgirdwood, 2019-05-06:
@bkokoszx I think it's always safer to do the initial copy to local memory to avoid races with the driver when using the shared mailbox.

bkokoszx, 2019-05-07:
@lgirdwood I agree that it is safer to do initial copy and use local variable, but after all we still invoke specific function with global context variable ```err = pipeline_params(pcm_dev->cd->pipeline, pcm_dev->cd,(struct sof_ipc_pcm_params *)_ipc->comp_data);``` or after #1336 I guess it will be something like ```err = pipeline_params(pcm_dev->cd->pipeline, pcm_dev->cd,(struct sof_ipc_pcm_params *)sof->ipc->comp_data);```. So my main concern is that we make a copy, but at the end we still refer to global variable ```_ipc``` or a member in sof struct ```sof->ipc``` (after #1336).

lgirdwood, 2019-05-08:
@bkokoszx _ipc is going in v1.4, it's WIP. Feel free to work on #1336 more as I'm travelling soon and want be able to do anything with it until at least after v1.3

jajanusz, 2019-06-07:
@bkokoszx I hope @lgirdwood answered your questions. Looks like issue is resolved for me. If you need more info feel free to reopen.

closed 2019-06-07 by jajanusz