thesofproject/sof #1340: [FEATURE] New Component to copy simultaneously to multiple sink buffers

ranj063, 2019-04-26:
**Is your feature request related to a problem? Please describe.**
This feature is required for BYT-CR where SSP0 and SSP2 use different AIFs

**Describe the solution you'd like**
A new component that can copy simultaneously to two sink buffers simultaneously. If one of the interfaces does not have a codec connected the samples are just dropped and should not cause an error. 

**Additional context**
Add any other context or screenshots about the feature request here.

ranj063, 2019-04-26:
@plbossart could you please add any information I missed here?

lgirdwood, 2019-04-29:
@ranj063 how does FW know which DAI has the codec connected ?

jajanusz, 2019-06-07:
@ranj063 can you do this with recent implementations of f.e. mux or you still need some specific component? I'm not sure how complete topology you'd like to look.

lgirdwood, 2019-06-07:
The only problem with this is it wastes a DMA channel. @ranj063 what's the usecase for this again, I guess this where we have some machines where the ASoC machine driver cant determine which SSP is used ?

kv2019i, 2025-12-17:
No activity in 5 years, closing. Please reopen if still relevant.