thesofproject/sof #1352: [BUG]Media playback is not at 48KHz

ranj063, 2019-05-01:
**Describe the bug**
Media pipeline playback is not at 48KHz. 

**To Reproduce**
aplay -Dhw:0,4 music.wav

**Expected behavior**
Playback should be 48KHz but what I see is 44.1KHz.

**Impact**
Showstopper

**Environment**
Kernel: sof-dev: https://github.com/thesofproject/linux/commit/a5ea42b5fddf8688ab541f5c6e9f425de5a8f295
FW: master: https://github.com/thesofproject/sof/commit/6e4faef8e7fea2e4ec71d9bc012680b446c7f8e5
Topology: sof-apl-pcm512x.tplg

4) Reproducibility Rate. 100%

ranj063, 2019-05-01:
@singalsu I get a feeling this has to do with topology. Could you please take a look at the sof-apl-pcm512x topology?

lgirdwood, 2019-05-01:
@ranj063 can you provide the output from aplay that shows it's 44.1kHz, you could try adding the -v flag to apaly too as that will dump buffers and other ALSA state. The SRC should work with 48 -> 48.

singalsu, 2019-05-02:
Thanks for reporting, I can do some quick checks (check playback length vs. assumed with "time aplay", check captured back 1 kHz tone frequency).

singalsu, 2019-05-02:
@ranj063 I wasn't able to repeat this with fresh git fetch of everything on my UP2. I checked from log that SRC with this topology was initialized for 48k-48k conversion and that playback of resampled to 48 kHz song of 3:44 long song took the correct amount of seconds with "time aplay" reported value.

Can you check from sof-logger output how SRC got initialized. In my case it was:

```
    0      2          SRC             56319.739583         3.854167      src/audio/src.c:594 	src_params()
    0      2          SRC             56323.072917         3.333333      src/audio/src.c:621 	src_params(), source_rate = 48000, sink_rate = 48000
    0      2          SRC             56325.937500         2.864583      src/audio/src.c:623 	src_params(), params->channels = 2, dev->frames = 48
    0      2          SRC             56330.104167         4.166667      src/audio/src.c:633 	src_params(), blk_in = 0, blk_out = 0

```

singalsu, 2019-05-02:
@ranj063 Though I spotted another bug with this topology. The buffers are not large enough for SRC to initialize with S24_4LE or S32_LE data format.

tlauda, 2019-06-05:
@ranj063 Is this still a valid case? It looks like @singalsu cannot reproduce it.

jajanusz, 2019-06-07:
@ranj063 This may no longer be bugged as @singalsu did some fixes to SRC, can you try to reproduce it with current codebase?

jajanusz, 2019-06-10:
@ranj063 Closing as others couldn't reproduce it. If you still have this issue feel free to reopen.

closed 2019-06-10 by jajanusz