thesofproject/sof #1364: [APL UP2 PCM512x] - [equlizer]: 'write error: Input/output error' occurred when tested the eq config files.

stevyan, 2019-05-05:
**Describe the bug**
'write error: Input/output error' occurred when tested the eq config files, and the failed config files perhaps are different when executed many times;

**To Reproduce**
1. cd <sof>/tools/eqctl
2. sof-eqctl -Dhw:0 -n 22 -s <all eq_iir_* files one by one>
3. sof-eqctl -Dhw:0 -n 23 -s <all eq_fir* files one by one>

**Actual Result**
```
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: pcm_write:2011: write error: Input/output error
Equalizer test failure with /home/ubuntu/sofcitest/./sof-test-suites/eq-test/eqctl/eq_fir_loudness.txt
```

**Environment**
date: 2019-5-05
linux (topic/sof-dev): 1071743a
sof (master): de5bba49 
topology: sof-apl-eq-pcm512x.tplg

**Screenshots or console output**
```
[    3.276620 <    0.004313>] mmcblk0: mmc0:0001 HBG4a2 29.1 GiB 
[    3.277147 <    0.000527>] mmcblk0boot0: mmc0:0001 HBG4a2 partition 1 4.00 MiB
[    3.277606 <    0.000459>] mmcblk0boot1: mmc0:0001 HBG4a2 partition 2 4.00 MiB
[    3.277827 <    0.000221>] mmcblk0rpmb: mmc0:0001 HBG4a2 partition 3 4.00 MiB, chardev (245:0)
[    3.279982 <    0.002155>]  mmcblk0: p1 p2
[    3.363405 <    0.083423>] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    3.486903 <    0.123498>] usb 1-2: new low-speed USB device number 2 using xhci_hcd
[    3.619780 <    0.132877>] usb 1-2: New USB device found, idVendor=046d, idProduct=c534, bcdDevice= 1.01
[    3.619787 <    0.000007>] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.628100 <    0.008313>] input: HID 046d:c534 as /devices/pci0000:00/0000:00:15.0/usb1/1-2/1-2:1.0/0003:046D:C534.0001/input/input3
[    3.680746 <    0.052646>] hid-generic 0003:046D:C534.0001: input,hidraw0: USB HID v1.00 Keyboard [HID 046d:c534] on usb-0000:00:15.0-2/input0
[    3.687269 <    0.006523>] input: HID 046d:c534 Mouse as /devices/pci0000:00/0000:00:15.0/usb1/1-2/1-2:1.1/0003:046D:C534.0002/input/input4
[    3.687633 <    0.000364>] hid-generic 0003:046D:C534.0002: input,hidraw1: USB HID v1.11 Mouse [HID 046d:c534] on usb-0000:00:15.0-2/input1
[    3.718909 <    0.031276>] systemd[1]: systemd 237 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN -PCRE2 default-hierarchy=hybrid)
[    3.731732 <    0.012823>] systemd[1]: Detected architecture x86-64.
[    3.735909 <    0.004177>] systemd[1]: Set hostname to <sh-apl-up2-pcm512x-05>.
[    3.776123 <    0.040214>] systemd-gpt-auto-generator[1435]: Failed to dissect: Input/output error
[    3.778925 <    0.002802>] systemd[1427]: /lib/systemd/system-generators/systemd-gpt-auto-generator failed with exit status 1.
[    3.794119 <    0.015194>] systemd[1]: File /lib/systemd/system/systemd-journald.service:36 configures an IP firewall (IPAddressDeny=any), but the local system does not support BPF/cgroup based firewalling.
[    3.794124 <    0.000005>] systemd[1]: Proceeding WITHOUT firewalling in effect! (This warning is only shown for the first loaded unit using IP firewalling.)
[    3.802907 <    0.008783>] usb 1-5: new full-speed USB device number 3 using xhci_hcd
[    3.881994 <    0.079087>] random: systemd: uninitialized urandom read (16 bytes read)
[    3.882034 <    0.000040>] systemd[1]: Reached target User and Group Name Lookups.
[    3.882305 <    0.000271>] random: systemd: uninitialized urandom read (16 bytes read)
[    3.882593 <    0.000288>] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[    3.882621 <    0.000028>] random: systemd: uninitialized urandom read (16 bytes read)
[    3.882688 <    0.000067>] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[    3.882863 <    0.000175>] systemd[1]: Created slice User and Session Slice.
[    3.909303 <    0.026440>] EXT4-fs (mmcblk0p2): re-mounted. Opts: errors=remount-ro
[    3.930604 <    0.021301>] usb 1-5: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
[    3.930607 <    0.000003>] usb 1-5: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.978560 <    0.047953>] Adding 1392256k swap on /swapfile.  Priority:-2 extents:5 across:1425024k SS
[    4.053682 <    0.075122>] systemd-journald[1454]: Received request to flush runtime journal from PID 1
[    4.224283 <    0.170601>] sof-audio-pci 0000:00:0e.0: PCI DSP detected
[    4.224547 <    0.000264>] sof-audio-pci 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.224634 <    0.000087>] sof-audio-pci 0000:00:0e.0: checking for capabilities at offset 0xc00
[    4.224636 <    0.000002>] sof-audio-pci 0000:00:0e.0: found ML capability at 0xc00
[    4.224639 <    0.000003>] sof-audio-pci 0000:00:0e.0: checking for capabilities at offset 0x800
[    4.224641 <    0.000002>] sof-audio-pci 0000:00:0e.0: found DSP capability at 0x800
[    4.224644 <    0.000003>] sof-audio-pci 0000:00:0e.0: checking for capabilities at offset 0x500
[    4.224646 <    0.000002>] sof-audio-pci 0000:00:0e.0: found GTS capability at 0x500
[    4.224648 <    0.000002>] sof-audio-pci 0000:00:0e.0: checking for capabilities at offset 0x1f00
[    4.224650 <    0.000002>] sof-audio-pci 0000:00:0e.0: found DRSM capability at 0x1f00
[    4.224653 <    0.000003>] sof-audio-pci 0000:00:0e.0: checking for capabilities at offset 0x700
[    4.224655 <    0.000002>] sof-audio-pci 0000:00:0e.0: found SPIB capability at 0x700
[    4.224667 <    0.000012>] sof-audio-pci 0000:00:0e.0: DMA mask is 64 bit
[    4.224671 <    0.000004>] sof-audio-pci 0000:00:0e.0: hda global caps = 0x1006701
```
[dmesg.txt](https://github.com/thesofproject/sof/files/3145407/dmesg.txt)

singalsu, 2019-05-06:
@stevyan The firmware needs to be built with xt-xcc to set FIR EQ to anything but pass-through. The gcc generated filter code is too slow to execute. Can you check with xt-xcc build this too.

singalsu, 2019-05-08:
@stevyan Can you please check if my above thought is correct. We can close this if IIR works with gcc & xt-xcc builds and FIR works with xt-xcc build.

It's known that any longer FIR filter execute fails with gcc build because the generated code is not good.

stevyan, 2019-05-08:
@singalsu it is occurred only for gcc compiler, and xcc compiled binary is ok.

singalsu, 2019-05-08:
OK, thanks! Let's close this then. I hope gcc becomes eventually better for xtensa.

closed 2019-05-08 by singalsu