thesofproject/sof #1370: [BUG] [APL] -22 after cycling D0 -> D3 -> D0 couple of times in monkey tests

smakows, 2019-05-06:
### Description

During running monkey tests in slim driver, after power cycling D0 -> D3 -> D0 couple of times with context saving and restoring, driver receives -22 from firmware on pipeline recreation, in phase of recreating pipeline streams.

### Recipe
FW commit de5bba49aaf2c0060a279de09580ac4205f550cb

Topology based on parallel internal SSP loopbacks

### Reproduction steps

1. Run monkey tests on APL machine
2. Wait till the test fail with -22 error response on pipeline recreation after power cycling

### Logs and files

[08_00_TestMonkeyRandomStreamOperations.zip](https://github.com/thesofproject/sof/files/3148242/08_00_TestMonkeyRandomStreamOperations.zip)

lgirdwood, 2019-05-06:
@smakows nice graphs :) Can you paste the failure log excerpt where this fails as it not easy to find the correct log in your attachment. Thanks.

smakows, 2019-05-07:
Overall test flow logs can be found in log.log_groups.log file
More detailed logs (with sent and received IPCs specified) can be found in log.log
Traces from consecutive fw lives (between power cycles) can be found in dma_trace_X.log files
Traces from last, failing fw life can be found in dma_trace_6.log file (6th lifecycle)

lgirdwood, 2019-05-08:
@smakows thanks, but we need the excerpt pasted otherwise it's difficult to discuss this issue without searching through the logs every time there is a comment. Just need about 10 - 20 lines of trace leading upto and including the issue.

smakows, 2019-05-09:
Fixed by fixing resource management mechanisms (freeing iob for traces in slim driver)

closed 2019-05-09 by smakows