thesofproject/sof #1374: [BUG] DSP oops information is not aligned between kernel and firmware

plbossart, 2019-05-06:
kernel xtensa.h:
/* Xtensa Firmware Oops data */
struct sof_ipc_dsp_oops_xtensa {
	struct sof_ipc_hdr hdr;

firmware xtensa.h

/* Xtensa Firmware Oops data */
struct sof_ipc_dsp_oops_xtensa {
	struct sof_ipc_dsp_oops_arch_hdr arch_hdr;
	struct sof_ipc_dsp_oops_plat_hdr plat_hdr;

Firmware folks, please fix this. I don't get how this ever worked.

wenqingfu, 2019-05-08:
@michalgrodzicki to reassign

jajanusz, 2019-05-10:
@plbossart @akloniex I see problem here: https://github.com/thesofproject/sof/commit/25cd7795cc1afa0eab6ac3e2fe96b43483f27297#diff-aade49d1736cebfadcd11a13ed068aafR42

We broke uapi headers here by including something from arch in uapi.
This needs to be done in a way that doesn't rely on arch-specific things like `XCHAL_NUM_AREGS`.

jajanusz, 2019-05-10:
There were some ideas according how it should be done https://github.com/thesofproject/sof/pull/591#discussion_r234635901 but looks like everybody forgot and kernel never used it as it should? - @mmaka1 @lgirdwood @plbossart Please figure out what you want to do with it.

plbossart, 2019-05-10:
@lgirdwood can you look into this, I don't have any context and you were the main reviewer on this one?

lgirdwood, 2019-05-11:
@plbossart will do, my understanding this update was internal after @akloniex comment. @akloniex can you update the C header in the ABI to match this PR data format change since it is actually external.

jajanusz, 2019-05-13:
@lgirdwood @plbossart I'll handle that with changing `uint32_t ar[XCHAL_NUM_AREGS];` to vararray, but you need to handle it on kernel side by reading `sof_ipc_dsp_oops_arch_hdr.totalsize` instead of `sizeof(sof_ipc_dsp_oops_xtensa)`

plbossart, 2019-05-13:
so we have no backwards compatible option here, right?

jajanusz, 2019-05-13:
@plbossart I can replace `XCHAL_NUM_AREGS` with something like `MAX_NUM_AREGS` and set it to highest from all platforms, but it'd be only temporary, final solution should work more like I proposed.

plbossart, 2019-05-14:
it looks like this never worked end-to-end, so we can claim there's no ABI update needed. Let's add this with the 'right' solution, and in the kernel we'll add this as a consolidation in MINOR6. 
@lgirdwood happy with this?

kv2019i, 2019-05-14:
@plbossart wrote:
> it looks like this never worked end-to-end, so we can claim there's no ABI update needed. Let's add 

I would go for this. The more wide-spread method for parsing dsp oopses has been relying on debugfs exception file that was not impacted by this (it used the sram window size info) and the info we have parsed for kernel message dumps has had the issues for many months, so this seems like this is now a bugfix (to reflect the headers to the real ABI).

closed 2019-05-16 by plbossart