thesofproject/sof #1375: [BUG] Detect test shall not be in uapi files

plbossart, 2019-05-06:
**Describe the bug**

sof_detect_test_config (detect_test.h) has no support in the kernel and shall not be in uapi files in the first place.
Please move somewhere else, or file a proper ABI change request.

wenqingfu, 2019-05-08:
@michalgrodzicki to reassign

jajanusz, 2019-05-10:
@plbossart It has no support in kernel but it was added+approved there https://github.com/thesofproject/sof/pull/1036 and ABI MINOR was increased, I think it was also before we agreed how to managed ABI changes. It is implemented in FW, struct is needed in uapi cos it's send via IPC, it works with slim driver, just kernel missed this feature.

plbossart, 2019-05-10:
I don't see a reason why you have an IPC that is only relevant for the slim driver. This makes no sense. Either it's a fully supported feature exposed in the right way or it's not.

jajanusz, 2019-05-10:
I just mentioned it works already with silm dirver and only kernel is behind. You can use it afaik.

plbossart, 2019-05-10:
use it to do what? what's the feature here? what are the topology tokens?
if you have a feature that's only tested with the slim driver, is this really a feature we support?

slawblauciak, 2019-05-10:
The structure is used to configure the detect_test component with certain detection parameters.
It's currently passed in the topology directly:
https://github.com/thesofproject/sof/blob/master/tools/topology/m4/detect_test_coef.m4

plbossart, 2019-05-10:
so is this like the eq parameters, something that is known by the apps but is just a generic IPC with a structure that's not known to the kernel?
When you say it's tested with the slim driver, I think we are blurring layers. It's rather the application using the slim driver that sets those values, yes?

jajanusz, 2019-05-10:
Slim driver (+python scripts) has to use the same api as kernel, it uses the same IPCs/structs to run FW. Just look at it as simple, slow driver that reloads full fw for every testcase (unless there is some specific scenario to have hot boot). So if it works with slim driver you should be able to use the same functionality from the "normal" driver.

slawblauciak, 2019-05-13:
> so is this like the eq parameters, something that is known by the apps but is just a generic IPC with a structure that's not known to the kernel?

Exactly

plbossart, 2019-05-13:
ok, so we can ignore this at the kernel level.
But we should still have an app which makes use of this component. @ranj063 @keyonjie what's missing here to perform the same tests with the kernel as with the slim driver?

xiulipan, 2019-05-13:
@keyonjie are we using `sof_eqctl` now to send `sof_detect_test_confi` to key_dectection?

keyonjie, 2019-05-14:
@plbossart @xiulipan yes, we are using sof_eqctl tool ATM to send detector config params, the config struct itself is transparent to kernel, but it need to be known by sof_eqctl or topology writer or any other apps who want to configure that.

xiulipan, 2019-05-14:
@keyonjie @lgirdwood @plbossart 
So we can say that this header in uapi is needed for users of SOF FW but not needed in the kernel, right?

plbossart, 2019-05-14:
ok so we will ignore this file in the kernel so closing this bug.
@keyonjie and team, you need to fix the sof_eqctl application. It has officially nothing specific to EQ since you use it for other purposes.

closed 2019-05-14 by plbossart