thesofproject/sof #1376: [BUG] SOF Manifest change is not tracked with ABI

plbossart, 2019-05-06:
**Describe the bug**

kernel manifest.h:
#define SOF_MAN_FW_HDR_FEATURES		0xff

firmware manifest.h:
#define SOF_MAN_FW_HDR_FEATURES		0x1ff

Please clarify if this is an ABI change, and if so get a proper ABI minor version. In any case please file a kernel PR, or if this is not necessary for userspace and kernel to know move out of uapi files.

jajanusz, 2019-05-07:
It's not ABI change, at least not for kernel, it's only used by the rimage tool to create proper manifest. Probably this file shouldn't even be in ABI...

jajanusz, 2019-05-07:
@plbossart @lgirdwood Architectural question - what should be in our `uapi`? 
Problem with this issue and similar is that we share something with 2 types of clients kernel/driver and userspace apps. In kernel uapi is for userspace apps, but will everything from our uapi also go to uapi of kernel or just part of it (so it is kinda just interface for kernel and for userspace should be somewhere else) ? 

Maybe we should have kernel<->FW ABI in `uapi` and headers that are shared for FW and apps (like rimage) in another folder (or just assume `include/sof` is public for these apps) ? I just want to be sure which folder are for exposing stuff for which consumer.

plbossart, 2019-05-07:
@jajanusz @lgirdwood 
It's a good question. If I look at the kernel files, the following files are NOT used:

include/uapi/sound/sof/manifest.h
include/uapi/sound/sof/trace.h
include/uapi/sound/sof/eq.h
include/uapi/sound/sof/tone.h

I would tend to agree that when the kernel does not need a file, we should not have those files in the kernel tree at all. if manifest.h and trace.h are only needed for rimage/sof-logger, then it's something out-of-scope. If we keep them in the tree we should agree there is no ABI tracking.

I am not sure also if we care about the EQ and tone at the kernel level either (these two components are not fully supported)? This doesn't scale, we are not going to add effect-specific header files for each new module.

wenqingfu, 2019-05-08:
@juimonen Would you please take this?

juimonen, 2019-05-09:
@wenqingfu ACK

@plbossart @jajanusz @lgirdwood if everybody agrees of removing these from kernel, the PR is:
https://github.com/thesofproject/linux/pull/923

I talked with @singalsu and at least removing eq headers should not be an issue..

plbossart, 2019-05-09:
fine with me.

plbossart, 2019-05-09:
solved by https://github.com/thesofproject/linux/pull/923

closed 2019-05-09 by plbossart