thesofproject/sof #1378: [BUG] UAPI tokens.h not aligned between kernel and firmware

plbossart, 2019-05-06:
**Describe the bug**

The SOF repo did not follow the kernel changes for the effect->process change, please fix. we should not have any EFFECT_TYPE anywhere in the firmware, it's PROCESS_TYPE now.

Also the definitions for SCHED_PERIOD/DEADLINE and TIME_DOMAIN/TIMER were not updated.

What a glorious mess.

wenqingfu, 2019-05-08:
@michalgrodzicki to reassign

plbossart, 2019-05-09:
@juimonen can you chime and check with @jajanusz what needs to be done?

juimonen, 2019-05-09:
@plbossart @jajanusz 
this should do it, but please you check it also:
https://github.com/thesofproject/sof/pull/1402

as I see these are defined correctly in tokens.m4 which topology uses.

jajanusz, 2019-05-10:
WIP #1381

PR #1381 (2019-05-12): uapi: token: minor change to align with topology configuration
This won't introduce ABI bump, only alignment with driver side header
file. Aim to fix #1378 

Reported-by: Keqiao Zhang <keqiao.zhang@linux.intel.com>
closed 2019-05-12 by tlauda