thesofproject/sof #1379: [BUG] UAPI trace.h is not aligned between kernel and firmware.

plbossart, 2019-05-06:
**Describe the bug**

trace.h contains a slew of definitions that are not used by the kernel. Either they are needed and we should have a proper ABI change, or they are not and the firmware needs to declared all this in another header.

````c
/* trace event classes - high 8 bits*/
#define TRACE_CLASS_IRQ		(1 << 24)
#define TRACE_CLASS_IPC		(2 << 24)
#define TRACE_CLASS_PIPE	(3 << 24)
#define TRACE_CLASS_HOST	(4 << 24)
#define TRACE_CLASS_DAI		(5 << 24)
#define TRACE_CLASS_DMA		(6 << 24)
#define TRACE_CLASS_SSP		(7 << 24)
#define TRACE_CLASS_COMP	(8 << 24)
#define TRACE_CLASS_WAIT	(9 << 24)
#define TRACE_CLASS_LOCK	(10 << 24)
#define TRACE_CLASS_MEM		(11 << 24)
#define TRACE_CLASS_MIXER	(12 << 24)
#define TRACE_CLASS_BUFFER	(13 << 24)
#define TRACE_CLASS_VOLUME	(14 << 24)
#define TRACE_CLASS_SWITCH	(15 << 24)
#define TRACE_CLASS_MUX		(16 << 24)
#define TRACE_CLASS_SRC         (17 << 24)
#define TRACE_CLASS_TONE        (18 << 24)git 
#define TRACE_CLASS_EQ_FIR      (19 << 24)
#define TRACE_CLASS_EQ_IIR      (20 << 24)
#define TRACE_CLASS_SA		(21 << 24)
#define TRACE_CLASS_DMIC	(22 << 24)
#define TRACE_CLASS_POWER	(23 << 24)
#define TRACE_CLASS_IDC		(24 << 24)
#define TRACE_CLASS_CPU		(25 << 24)
#define TRACE_CLASS_EDF		(27 << 24)
#define TRACE_CLASS_KPB		(28 << 24)
#define TRACE_CLASS_SELECTOR	(29 << 24)
#define TRACE_CLASS_SCHEDULE	(30 << 24)
#define TRACE_CLASS_SCHEDULE_LL	(31 << 24)
````

wenqingfu, 2019-05-08:
@michalgrodzicki to reassign

plbossart, 2019-05-09:
https://github.com/thesofproject/linux/pull/923 suggests to remove this from kernel files.
Agreed?

plbossart, 2019-05-09:
solved by https://github.com/thesofproject/linux/pull/923

closed 2019-05-09 by plbossart