thesofproject/sof #1380: [BUG] UAPI topology.h file is not aligned between kernel and firmware

plbossart, 2019-05-06:
**Describe the bug**

the definitions for sof_ipc_process_type is not included/used at the firmware level. How can this possibly work since comp_process will pass a 'type'? The natural conclusion is that the process_type is not checked for illegal values?

sof_event_types are defined for the kernel but not for firmware so either they are needed by the firmware or the kernel should be the only one dealing with those definitions.

wenqingfu, 2019-05-08:
@michalgrodzicki to reassign

plbossart, 2019-05-09:
@juimonen can you chime in, you were involved in the effect->process change.

juimonen, 2019-05-10:
@plbossart: was not involved in it too much, only barking...

1) so in user space the topology includes PROCESS_TYPE token which is a string, like "MUX".
2) in kernel we have a mapping array from this string to sof_comp_type, uint32 (topology.h)
3) in firmware the component is registering itself into a global list with sof_comp_type (topology.h)
4) in component creation the comp type (uint32) coming from kernel is search from the list of registered components

So definitely you can go wrong if the topology.h is not aligned (the id/enum is different in both sides). 
I'm not sure how you could check this...? I mean if you shuffle the enum ordering in topology.h, things will go bad. This is what I saw when fileread enums we're moved from the middle to the end of enum definition.

juimonen, 2019-05-10:
@plbossart so why it is working is that topology/kernel maps the SOF_PROCESS_TYPE to sof_comp_type, but firmware is not using SOF_PROCESS_TYPE at all.

juimonen, 2019-05-10:
https://github.com/thesofproject/sof/pull/1404

plbossart, 2019-05-10:
As commented on #1404, this issue masks a problem. We have to stop using enums for process_type.

plbossart, 2019-05-14:
I am not going to lay on the tracks on this one since it's not really an ABI issue but rather an miss on how we do component integration. It's likely better to file another issue to track this for 1.4.

jajanusz, 2019-05-16:
Closing, because we agreed to solve root of the problem (definitions) later and then maybe align API, but it's going to be another issue.

closed 2019-05-16 by jajanusz