thesofproject/sof #1382: [BUG]selector buffer new fails with KWD integrated topology on GLK

keyonjie, 2019-05-07:
**Describe the bug**
selector buffer new fails with KWD integrated topology on GLK.

**To Reproduce**
Just rebooting, and capture dmesg and sof-logger logs.

**Expected behavior**
PCM created and sound card created.

**Impact**
Influence KWD feature working on GLK.

**Environment**
1) Kernel:
https://github.com/thesofproject/linux/tree/topic/keyword_detect
(only one commit newer on top of topic/sof-dev)
FW:
https://github.com/thesofproject/sof
(+ PR #1350 #1362 #1365)
2) topology file: sof-glk-da7219-kwd.tplg
3) GLK ubuntu
4) 100%.

**dmesg & sof-logger logs**

```
[    3.486524] sof-audio-pci 0000:00:0e.0: ASoC: creating DAPM widget BUF9.1 id 16
[    3.486528] sof-audio-pci 0000:00:0e.0: tplg: ready widget id 59 pipe 9 type 31 name : BUF9.1 stream none
[    3.486530] sof-audio-pci 0000:00:0e.0: buffer BUF9.1: size 2560 caps 0x71
[    3.486533] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    3.486621] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    3.486624] sof-audio-pci 0000:00:0e.0: ASoC: creating DAPM widget BUF9.2 id 16
[    3.486626] sof-audio-pci 0000:00:0e.0: tplg: ready widget id 60 pipe 9 type 31 name : BUF9.2 stream none
[    3.486628] sof-audio-pci 0000:00:0e.0: buffer BUF9.2: size 2560 caps 0x71
[    3.486630] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[    3.486728] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x30200000 size 20
[    3.486730] sof-audio-pci 0000:00:0e.0: error: buffer BUF9.2 load failed
[    3.486732] sof-audio-pci 0000:00:0e.0: error: DSP failed to add widget id 0 type 31 name : BUF9.2 stream none reply -12
[    3.486734] sof-audio-pci 0000:00:0e.0: ASoC: failed to load widget BUF9.2
[    3.486761] sof-audio-pci 0000:00:0e.0: error: tplg component load failed -12
[    3.486769] sof-audio-pci 0000:00:0e.0: error: failed to load DSP topology -22
[    3.486771] sof-audio-pci 0000:00:0e.0: ASoC: failed to probe component -22
[    3.486794] bxt_da7219_max98357a glk_da7219_max98357a: ASoC: failed to instantiate card -22
[    3.486834] bxt_da7219_max98357a: probe of glk_da7219_max98357a failed with error -22
[    3.634241] random: crng init done
[    3.634243] random: 7 urandom warning(s) missed due to ratelimiting
```

```
root@test-Bobba:~# ./log_e 
    0      2          IPC          21156196.822917         3.802083    src/ipc/handler.c:951 	ipc: pipe 8 comp 51 -> new (type 16)
    0      2          KPB          21156216.354167        19.531250      src/audio/kpb.c:94  	kpb_new()
    0      2          KPB          21156225.052083         8.697917      src/audio/kpb.c:186 	kpb_allocate_history_buffer()
    0      1          MEM          21156245.208333        20.156250      src/lib/alloc.c:271 	error: 342 blocks needed for allocation but only 341 blocks are remaining
    0      2          KPB          21156257.968750        12.760417      src/audio/kpb.c:212 	kpb new memory block: 130816
    0      2          KPB          21156264.062500         6.093750      src/audio/kpb.c:212 	kpb new memory block: 3584
    0      2          KPB          21156267.395833         3.333333      src/audio/kpb.c:259 	kpb_allocate_history_buffer(): allocated 134400 bytes
    0      2          IPC          21156335.052083        67.656250    src/ipc/handler.c:981 	ipc: pipe 8 buffer 52 -> new (0xa00 bytes)
    0      2       BUFFER          21156353.802083        18.750000   src/audio/buffer.c:54  	buffer_new()
    0      2          IPC          21156436.354167        82.552086    src/ipc/handler.c:981 	ipc: pipe 8 buffer 53 -> new (0xa00 bytes)
    0      2       BUFFER          21156455.468750        19.114584   src/audio/buffer.c:54  	buffer_new()
    0      2          IPC          21156536.822917        81.354164    src/ipc/handler.c:948 	ipc: hdr 0x30010000 tx (80) > rx (28)
    0      2          IPC          21156540.572917         3.750000    src/ipc/handler.c:951 	ipc: pipe 8 comp 54 -> new (type 2)
    0      2          DAI          21156560.989583        20.416666      src/audio/dai.c:194 	dai_new()
    0      2         DMIC          21156569.843750         8.854167 rs/intel/cavs/dmic.c:1523	dmic_probe()
    0      2        POWER          21156578.333333         8.489583 el/cavs/pm_runtime.c:122 	dis-dmic-clk-gating index 1 CLKCTL 05600002
    0      2          DAI          21156583.020833         4.687500        src/lib/dai.c:87  	dai_get(), d = 0xbe044088, sref = 1
    0      2          DMA          21156587.447917         4.427083        src/lib/dma.c:142 	dma_get() ID 0 sref = 5 busy channels 0
    0      2          IPC          21156728.697917       141.250000    src/ipc/handler.c:1009	ipc: pipe 8 -> new
    0      2         PIPE          21156765.000000        36.302082 src/audio/pipeline.c:72  	pipeline_new()
    0      2          IPC          21156862.656250        97.656250    src/ipc/handler.c:948 	ipc: hdr 0x30010000 tx (220) > rx (28)
    0      2          IPC          21156866.354167         3.697917    src/ipc/handler.c:951 	ipc: pipe 9 comp 57 -> new (type 15)
    0      2      unknown          21156886.927083        20.572916 /audio/detect_test.c:211 	test_keyword_new()
    0      2          IPC          21156956.718750        69.791664    src/ipc/handler.c:948 	ipc: hdr 0x30010000 tx (200) > rx (28)
    0      2          IPC          21156960.625000         3.906250    src/ipc/handler.c:951 	ipc: pipe 9 comp 58 -> new (type 17)
    0      2     SELECTOR          21156981.562500        20.937500 src/audio/selector.c:115 	selector_new()
    0      2          IPC          21157053.697917        72.135414    src/ipc/handler.c:981 	ipc: pipe 9 buffer 59 -> new (0xa00 bytes)
    0      2       BUFFER          21157074.739583        21.041666   src/audio/buffer.c:54  	buffer_new()
    0      2          IPC          21157150.677083        75.937500    src/ipc/handler.c:981 	ipc: pipe 9 buffer 60 -> new (0xa00 bytes)
    0      2       BUFFER          21157172.395833        21.718750   src/audio/buffer.c:54  	buffer_new()
    0      1          MEM          21157179.427083         7.031250      src/lib/alloc.c:271 	error: 7 blocks needed for allocation but only 4 blocks are remaining
    0      1       BUFFER          21157188.125000         8.697917   src/audio/buffer.c:77  	buffer_new() error: could not alloc size = 2560 bytes of type = 113
    0      1          IPC          21157194.114583         5.989583        src/ipc/ipc.c:217 	ipc_buffer_new() error: buffer_new() failed
    0      1          IPC          21157199.375000         5.260417    src/ipc/handler.c:987 	ipc: pipe 9 buffer 0 creation failed 0

```

keyonjie, 2019-05-07:
@mrajwa I add this to you and hope you can look into it.

mrajwa, 2019-05-08:
@keyonjie I took a look at that and it looks like we run out of SOF_MEM_CAPS_RAM. I bet if you make a toplogy a little bit lighter by removing some pipeline for example the problem won't show up. Can you confirm that?

keyonjie, 2019-05-08:
> @keyonjie I took a look at that and it looks like we run out of SOF_MEM_CAPS_RAM. I bet if you make a toplogy a little bit lighter by removing some pipeline for example the problem won't show up. Can you confirm that?
@mrajwa yes, I confirm that, as i talked to you, it works when I remove the PCM99 pipeline.

mrajwa, 2019-05-08:
@keyonjie, please check if this patch #1394 fixes the issue on your side.

PR #1399 (2019-05-09): Topology: kwd: change to use correct buffer caps
This is aimed to fix #1382 

We don't need to use host/dai caps for detector buffer and selector
buffer, here correct it.
closed 2019-05-09 by tlauda