thesofproject/sof #1385: [BUG] Rimage reports errors in correct image

jajanusz, 2019-05-07:
**Describe the bug**
While doing typical build for APL. I get part of log like:
```
Module Reading bootloader-apl
error: can't find section .bss in module bootloader-apl
 BSS module metadata section at index -22
error: can't find section .static_log_entries in module bootloader-apl
 static log entries section at index -22
error: can't find section .fw_ready in module bootloader-apl
 fw ready section at index -22 
```

**To Reproduce**
Build default config for apollolake.

**Expected behavior**
It shouldn't be an error. It may be warning, or no message at all if rimage would know what it should have in bootloader/fw parts of images. 

**Impact**
Annoyance - we often get log messages pointing to that part of log as *error* when there is no error or it is later in log.

lgirdwood, 2019-05-08:
@jajanusz ack, should probably be an INFO level message since all 3 sections are optional.

jajanusz, 2019-05-10:
@lgirdwood I thought it depends on how you use rimage. For FW that is made from 1 file it has to have all these sections, but when bootloader is in seperate file then some sections are there and rest are in main binary. So it should be error for 1 file and no error when sections are just in another files.

lgirdwood, 2019-05-11:
@jajanusz ok make sense, fwiw rimage can take multiple ELFs as args and your right basefw would need to generate errors but this would be info for fw runtime modules.

PR #1542 (2019-06-11): rimage: ignore sections not used by bootloader
Fixes #1385 

In case of multiple modules, rimage should not expect these sections
in bootloader module: .bss, .static_log_entries .fw_ready.
closed 2019-06-11 by tlauda