thesofproject/sof #1386: [FEATURE] Volume: Support logarithmic gain ramps

singalsu, 2019-05-07:
Logaritmic gain ramps create subjectively equal amount of loudness change per unit of time so it could be the preferred ramp type. It also allows usage of shorter time ramps while avoiding clicks or zipper noise. Currently SOF only supports linear ramps. The topology and driver supports the ramp type topology token so it would be a FW only change.

The ramp type was omitted from recent PR for volume ramp. The implementation would need log() function to DSP. Such function to cover the volume range with is possible to compute via a Taylor series approximation/iteration done with integer fractional math.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lgirdwood, 2019-05-08:
@singalsu I dont think we need the precision of log() in the DSP for gain ramping, we most likely need a compressed log lookup table. i.e. log table uses same array element where log() output does not significantly change compared to previous or next input values. iiuc, we should only need to support a finite and known range of possible log() input values too ?

singalsu, 2025-12-17:
I think we can close this, we have the Windows Fade as a more sophisticated option over basic linear ramp.