thesofproject/sof #1388: [FEATURE]trace point reorder in platform_init function

xiulipan, 2019-05-07:
**Is your feature request related to a problem? Please describe.**
The trace point value now used in `platform_init` is out of order. After some refine of codec, there leave some blank trace point as well as miss trace point. The trace point is now misleading for new platform enable and QEMU debug.

**Describe the solution you'd like**
The trace point should be grouped and increase in order and did not miss any init function track.

bkokoszx, 2019-05-21:
@xiulipan 
Do you mean for example that TRACE_BOOT_SYS_SCHED (0x3500) appears after TRACE_BOOT_PLATFORM_CLOCK (0x4150)?

lgirdwood, 2019-05-22:
@bkokoszx yes, the ordering should be sequential with gaps for future use.

PR #1520 (2019-06-06): init: reorder trace points sequence
Reorders trace points sequence to be incremental.
Also adds additional trace points for better debug.

Fixes #1388.
closed 2019-06-06 by dbaluta