thesofproject/sof #1389: [FEATURE]trace point on APL+ platform is written into memory window

xiulipan, 2019-05-07:
**Is your feature request related to a problem? Please describe.**
trace point should be written into register to give a full round tracking.

**Describe the solution you'd like**
Change the trace point into ipc regs. May also need kernel driver change for the trace point for different platforms.

**Additional context**
Now trace point before memory window init or platform without memory window will have no trace point.
https://github.com/thesofproject/sof/blob/05d4e190d437da8e6be1478e52bd39c71a15df52/src/platform/intel/cavs/platform.c#L402-L422

tlauda, 2019-06-05:
@xiulipan I understand your thinking, but I don't see any spare registers to use for that and ipc regs are for sure not a good idea. Unless platform offers dedicated log regs I don't think we can use anything besides memory window. Anyway memory window with status regs is enabled by ROM, so we have traces even before FW boot. If some platform will ever have some dedicated regs we can revisit this. Are you OK to close?

xiulipan, 2019-06-05:
@tlauda 
https://github.com/thesofproject/sof/blob/eca023476fce5782873dfe1123ab68afe31493f0/src/arch/xtensa/boot_loader.c#L239-L240

I think this would be a TODO for APL+, which means from CNL and CNL+

tlauda, 2019-06-05:
@xiulipan It's pretty old comment. I know that on non-cAVS platforms trace points are in fact put into IPC regs, but I'm not fan of doing it also on cAVS platforms, since we have memory window.
@lgirdwood What's your take?

lgirdwood, 2019-06-05:
This is probably best via a shared register since it does not depend on SRAM windows init being complete, but not sure how easy this will be as IIUC the BUSY IRQ bit needs to be set in order for the updates to appear on the host side.

tlauda, 2019-06-11:
@lgirdwood For cAVS ROM enables memory window, so trace points should be available as soon as ROM jumps to FW.

lgirdwood, 2019-06-11:
@tlauda ok thanks, SRAM window it is then. @xiulipan  no need to use IPC register now, comment can be deleted.

PR #1553 (2019-06-12): boot_loader: remove deprecated comment
Removes deprecated comment from boot_loader.
Memory window is enabled by ROM even before
FW is loaded, so there is no need for usage
of additional registers.

Closes #1389.
closed 2019-06-12 by tlauda