thesofproject/sof #1391: [FEATURE] Allow updating EQs during playback/capture

singalsu, 2019-05-08:
Currently the EQs can be updated by sof-eqctl only when the pipeline they belong to is idle.

In audio tuning scenario it is desirable to on-the-fly re-configure the EQs to quickly measure or subjectively evaluate the impact of new tuning without need to start/stop system audio servers etc.

Therefore supporting the change in runtime when the resources (RAM) needed by new tuning are equal or less compared old new tuning the switch should be supported. An attempt to use more resources can be prevented if there's risk that buffers re-allocation consumes too much time.

Due to only targeting audio tuning engineer's needs this enhancement does not suggest a perfectly glitch-free transition. Especially the recursive filters need to be reset to ensure stable operation. Due to the state variables reset there will some small break in audio when re-configured.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

singalsu, 2019-05-08:
@lgirdwood @dbogdzix I just created this enhancement to address the closed Docs issue #116.

lgirdwood, 2019-08-30:
@singalsu I've made this P1 for v1.5, the EQ should accept the new coefficients and only update them when after copy() completes.
Btw, I'm assuming updating the coefficients here is equivalent of adjusting EQ sliders on a traditional HiFi ?

singalsu, 2019-09-02:
@lgirdwood Yep, the update on the fly can be made if the new configuration request is now larger than previous. It's not in practice a severe limitation, in filter tuning the IIR or FIR order can be kept the same. I'm assuming we would not do free()-malloc() during playback.

We don't have direct slider adjustable EQ type in firmware. But it could be supported by having some computing in user space. For IIR it would be quite simple (for parametric EQ or fixed bands "GUI") and for FIR (arbitrary response "GUI") need IFFT in user space.

lgirdwood, 2019-09-06:
@singalsu sounds like a topic for the audio conference. @plbossart I guess we need alsamixer/alsalib support for binary TLV sliders for EQ type users.

plbossart, 2019-09-06:
That wasn't what I was suggesting, rather that the application can provide new binary data at any time, and the the firmware takes care of smoothing transitions to avoid pop-clicks.

singalsu, 2019-11-05:
I've now started working with this feature.

singalsu, 2019-11-06:
SOF PR #2065 implements this feature for IIR EQ type. I'll do the same for FIR after the solution for IIR is reviewed.

singalsu, 2019-12-19:
Status update: This is half done. I still need to make this for FIR. ETA ww02 2020.

singalsu, 2020-01-08:
SOF PR #2259 will complete the work for this enhancement issue when it is merged.

singalsu, 2020-01-15:
This feature is complete since SOF commit c0b9b4057e4dc408c9af5c9695b9d052ab1d3702 .

singalsu, 2020-01-15:
This is OK to close!

closed 2020-01-15 by singalsu