thesofproject/sof #1392: [BUG] Tesbench file component is not timer based scheduling compliant

singalsu, 2019-05-08:
**Describe the bug**
The SRC test in testbench freezes with 48 kHz to 11.025 kHz test case. The fileread does not provide enough data for SRC. If SRC returns no data the pipeline tries to recover and re-prepare the pipeline but it does not help and the simulation freezes to repeat the pipeline copy and repeat.

**To Reproduce**
Run src test case in sof/tools/test/audio.

**Expected behavior**
Test script completes.

**Impact**
Not possible to test objective audio parameters for components.

**Environment**
1) Branch name and commit hash of 3 repositories: sof (firmware), linux (kernel driver) and soft (tools & topology).
2) Name of the topology file
3) Name of the platform(s) on which the bug is observed.
4) Reproducibility Rate. If you can only reproduce it randomly, it’s useful to report how many times the bug has been reproduced vs. the number of attempts it’s taken to reproduce the bug.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

jajanusz, 2019-05-10:
I can confirm, I also noticed that SRC with conversion to lower rates doesn't work. I can tell you that it didn't work even before I started fixing testbench, so I continued testing only with converstions to higher rates (just to see if I didin't break it more).

singalsu, 2019-05-10:
I've made a fix PR into #1409.

@jajanusz Yep, you are right it was broken before your work. Due to inclusion of volume into pipeline everything but 24 bit mode tests failed.

singalsu, 2019-05-10:
The problem with volume is that testbench does not set it up properly. I need to work with it next (can add gain and THD+N cases for it).

I wan't to add EQ test cases as well since currently there's no proper validation. I had the tests running in an ancient version of testbench that stopped being compatible so at the moment I can't change the processing arithmetic there without worry of some audio quality loss. It's hard to judge by ear the changes even if I have pretty good studio headphones at work for subjective assessment.

singalsu, 2019-05-17:
The PR with fix is merged so this can be closed.

closed 2019-05-17 by singalsu