thesofproject/sof #1396: [BUG][CML]IPC timed out for '0x60040000: GLB_STREAM_MSG: TRIG_START'

ClarexZhou, 2019-05-09:
**Summary:**
'ipc timed out for '0x60040000: GLB_STREAM_MSG: TRIG_START' after system reboot.

**To reproduce**
Do system reboot stress test. 

**Actual result**
error: ipc timed out for '0x60040000: GLB_STREAM_MSG: TRIG_START'

**Reproduce rate:** 1%

**Environment**
Rander UCM https://github.com/RanderWang/intel-UCM   
Kernel repo：c99938c https://github.com/thesofproject/linux/tree/release/sof-v5.0 
Firmware repo: 05d4e19  https://github.com/thesofproject/sof/commits/master |  
Topology: https://github.com/thesofproject/sof/commits/master . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

**error sof-logger**
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1       VOLUME           5334495.520833   5334495.500000   src/audio/volume.c:474 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           5334543.020833        47.500000 src/audio/pipeline.c:738 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6       5334549.687500         6.666667 src/audio/pipeline.c:846 	pipeline_xrun_recover()
    0      1       VOLUME           5338245.000000      3695.312500   src/audio/volume.c:474 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           5338291.718750        46.718750 src/audio/pipeline.c:738 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6       5338298.489583         6.770833 src/audio/pipeline.c:846 	pipeline_xrun_recover()
    0      1       VOLUME           5341994.687500      3696.197998   src/audio/volume.c:474 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           5342041.406250        46.718750 src/audio/pipeline.c:738 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6       5342048.020833         6.614583 src/audio/pipeline.c:846 	pipeline_xrun_recover()
    0      1       VOLUME           5345744.791667      3696.770752   src/audio/volume.c:474 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           5345791.250000        46.458332 src/audio/pipeline.c:738 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6       5345797.864583         6.614583 src/audio/pipeline.c:846 	pipeline_xrun_recover()
    0      1       VOLUME           5349494.583333      3696.718750   src/audio/volume.c:474 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           5349540.989583        46.406250 src/audio/pipeline.c:738 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6       5349547.864583         6.875000 src/audio/pipeline.c:846 	pipeline_xrun_recover()
    0      1       VOLUME           5353245.052083      3697.187500   src/audio/volume.
```
**dmesg**
```
[    6.368975] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 0 dir 0
[    6.368985] sof-audio-pci 0000:00:1f.3: generating page table for 00000000fc8aaece size 0xffc0 pages 16
[    6.369035] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[    6.369037] sof-audio-pci 0000:00:1f.3: periods:4
[    6.369066] sof-audio-pci 0000:00:1f.3: stream_tag 1
[    6.369075] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[    6.369319] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[    6.369328] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[    6.568048] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[    6.568080] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[    6.871913] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60040000 size 12
[    6.871953] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    6.872155] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[    6.872159] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[    6.872174]  HDA Analog: ASoC: trigger FE failed -110
```
[dmesg_failed_41.log](https://github.com/thesofproject/sof/files/3160235/dmesg_failed_41.log)
[logger-t_failed_41.log](https://github.com/thesofproject/sof/files/3160236/logger-t_failed_41.log)

tlauda, 2019-05-10:
FW logs indicate that HDA Link pointer is stuck after trigger start, so there is something wrong with the Link DMA on the kernel side.

ClarexZhou, 2019-05-14:
0x60040000: GLB_STREAM_MSG: TRIG_START timed out occurred on CFL-s during playback stress test after running for about 210 times
Error Sof-logger:
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         PIPE 1.6     385989754.531250 385989760.000000 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         385993446.145833      3691.614502 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         385993492.135417        45.989582   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         385993497.604167         5.468750 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     385993504.375000         6.770833 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         385997196.250000      3691.875000 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         385997242.135417        45.885418   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         385997247.604167         5.468750 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     385997254.375000         6.770833 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         386000946.093750      3691.718750 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         386000992.031250        45.937500   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         386000997.552083         5.520833 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     386001004.270833         6.718750 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         386004695.781250      3691.510498 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         386004741.770833        45.989582   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         386004747.239583         5.468750 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     386004753.802083         6.562500 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         386008445.937500      3692.135498 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         386008492.031250        46.093750   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         386008497.343750         5.312500 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     386008503.802083         6.458333 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         386012195.833333      3692.031250 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         386012240.677083        44.843750   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         386012245.989583         5.312500 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     386012252.552083         6.562500 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         386015946.875000      3694.322998 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         386015991.875000        45.000000   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         386015997.187500         5.312500 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     386016003.750000         6.562500 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1       VOLUME         198345994.062500              nan   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         198345999.635417         5.572917 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     198346006.354167         6.718750 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         210025947.187500  11679941.000000 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         210025994.166667        46.979168   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         210025999.531250         5.364583 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
[... cut ...]
Firmware repo: b73b9a6 https://github.com/thesofproject/sof/commits/master +PR1354  
Topology: https://github.com/thesofproject/sof/commits/master . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

[ipctimedoutAplay_dmesg.log](https://github.com/thesofproject/sof/files/3176061/ipctimedoutAplay_dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3176062/logger.log)
[errorlogger.log](https://github.com/thesofproject/sof/files/3176063/errorlogger.log)

tlauda, 2019-05-14:
HDA Link problem. Please verify with the same changes as in #1413.

wenqingfu, 2019-05-14:
with 1354 and 895 integrated (aka, the fixes mentioned in #1413), there is still occurrence

YvonneYang2, 2019-05-17:
Issue occurred once after do steps below:
1.System reboot with headphone connected.
2.Paplay 1.wav
3.Connect the external DP monitor, and choose to HDMI2/DP output setting in Sound settings -Output.
4.Switch the setting between headphone and HDMI2/DP2 in Sound settings -Output.

**Environment**
Rander UCM https://github.com/RanderWang/intel-UCM/tree/hdmi 
Kernel repo：df06e42 https://github.com/thesofproject/linux/tree/release/sof-v5.0
Firmware repo: b73b9a6 https://github.com/thesofproject/sof/commits/debug-ipc-timeout
Topology: https://github.com/thesofproject/sof/commits/debug-ipc-timeout . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg
platform: CML laptop with ALC3204 codec

**dmesg**
```
[  656.975512] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  656.975552] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.001369] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.001400] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.027676] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.027708] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.053661] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.053694] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.079718] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.079757] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.105711] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.105751] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.119239] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60010000 size 108
[  657.119267] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  657.119542] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  657.119558] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  657.119564] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[  657.119570] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
[  657.119584]  HDA Analog: ASoC: hw_params FE failed -110
[  657.119612] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 1
[  657.119619] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  657.131397] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.131437] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.157698] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.157738] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.183722] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.183761] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.209720] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.209760] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.235720] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.235760] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  657.261728] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  657.261768] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```
**sof-logger**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      2          DAI 2.13    141139694.531250 141139696.000000      src/audio/dai.c:519 	dai_comp_trigger(), START
    0      2          DMA         141140871.041667      1176.510376 intel/cavs/hda-dma.c:533 	hda-dmac: 6 channel 0 -> start
    0      2          DMA         141140875.416667         4.375000 intel/cavs/hda-dma.c:369 	hda-dmac: 6 channel 0 -> enable
    0      2       BUFFER         141142122.447917      1247.031250   src/audio/buffer.c:118 	comp_update_buffer_produce(), no bytes to produce
    0      1         COMP         141142127.343750         4.895833 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      2         COMP         141142132.656250         5.312500 of/audio/component.h:724 	comp_overrun(), ((dev->comp.id << 16) | sink->free) = 655360, ((min_bytes << 16) | copy_bytes) = 0
    0      2         PIPE 2.14    141142136.406250         3.750000 src/audio/pipeline.c:579 	pipeline_trigger()
    0      2          DAI 2.13    141142140.729167         4.322917      src/audio/dai.c:508 	dai_comp_trigger(), command = 8
    0      2          DAI 2.13    141142144.479167         3.750000      src/audio/dai.c:566 	dai_comp_trigger(), XRUN
    0      2          DAI 2.13    141142147.968750         3.489583      src/audio/dai.c:572 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         141142151.875000         3.906250 intel/cavs/hda-dma.c:627 	hda-dmac: 6 channel 0 -> stop
    0      1       VOLUME         141142158.906250         7.031250   src/audio/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         141142164.375000         5.468750 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14    141142170.885417         6.510417 src/audio/pipeline.c:843 	pipeline_xrun_recover()
[... cut ...]
    0      1         COMP         141149627.552083         5.000000 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      2         COMP         141149632.968750         5.416667 of/audio/component.h:724 	comp_overrun(), ((dev->comp.id << 16) | sink->free) = 655360, ((min_bytes << 16) | copy_bytes) = 0
```
[ipctimeout_dmesg.log](https://github.com/thesofproject/sof/files/3190456/ipctimeout_dmesg.log)
[logger_ipctimeout.log](https://github.com/thesofproject/sof/files/3190457/logger_ipctimeout.log)

YvonneYang2, 2019-05-22:
Issues occurred during 1000 times playback(aplay -d 1) stress test . 

**Environment**
Firmware: 5fa2426 https://github.com/thesofproject/sof/tree/master
Kernel: b58f448 https://github.com/thesofproject/linux/tree/release/sof-v5.0
Topology: file: tools/topology/sof-hda-generic.tplg, sof-hda-generic-2ch.tplg same as firmware
Kconfig: https://sof-ci-storage.sh.intel.com/daily/20190521/release/linux/topic/sof-dev/901cd2a5-2/config-daily-hda-20190521.txt
platform ：CML laptop 

**dmesg**
```
[  724.881344] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  724.881351] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[  724.887273] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  724.887291] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  725.191325] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60040000 size 12
[  725.191351] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  725.191686] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  725.191699] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[  725.191703] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  725.191711]  HDA Analog: ASoC: trigger FE failed -110
[  725.191722] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  725.191746] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  725.191971] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  725.191980] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  725.192021] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  725.192115] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  725.192146] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  725.681336] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  725.681369] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  725.799544] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  725.799555] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  726.103296] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[  726.103317] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  726.103576] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  726.103586] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  726.103590] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  726.103747] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  726.180963] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
```
**error logger**
```
    0      1         COMP         647945558.645833 647945536.000000 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         647945605.781250        47.135418 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         647945611.250000         5.468750 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     647945618.020833         6.770833 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         647949307.812500      3689.791748 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         647949352.760417        44.947918 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         647949358.020833         5.260417 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     647949364.635417         6.614583 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         647953057.864583      3693.229248 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         647953102.760417        44.895832 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         647953108.020833         5.260417 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     647953114.635417         6.614583 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         647956807.916667      3693.281250 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         647956852.812500        44.895832 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         647956858.177083         5.364583 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     647956864.635417         6.458333 src/audio/pipeline.c:843 	pipeline_xrun_recover()
    0      1         COMP         647960558.229167      3693.593750 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
[... cut ...]
    0      1         COMP         647968058.072917      3693.489502 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
```

[playback_failed_353th_error_logger.log](https://github.com/thesofproject/sof/files/3205870/playback_failed_353th_error_logger.log)
[playback_failed_353th_logger.log](https://github.com/thesofproject/sof/files/3205871/playback_failed_353th_logger.log)
[playback_dmesg.log](https://github.com/thesofproject/sof/files/3205872/playback_dmesg.log)

ClarexZhou, 2019-05-24:
Error occurred during S3 with playback stress test. 

**Environment**
Kernel: 9453da1 https://github.com/lyakh/linux/tree/ipc-timeout (with PR895)
FW:b527fd7 https://github.com/lyakh/sof/tree/ipc-timeout (with PR1354)
Topology:Same as above FW repo, file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg
kconfig:
unset CONFIG_SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE
unset CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE

[s3_test_14_dmesg.log](https://github.com/thesofproject/sof/files/3215087/s3_test_14_dmesg.log)
[s3_test_14_logger_error.log](https://github.com/thesofproject/sof/files/3215088/s3_test_14_logger_error.log)

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         COMP         172202696.770833 172202704.000000 rc/audio/component.c:140 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_START
    0      1          IPC         172202701.302083         4.531250 src/audio/pipeline.c:606 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 1
    0      1          IPC         172202706.406250         5.104167    src/ipc/handler.c:513 	ipc: comp 1 trigger 0x40000 failed -22
    0      1         COMP         172580911.458333    378205.062500 rc/audio/component.c:161 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC         172580915.572917         4.114583 src/audio/pipeline.c:606 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 0
    0      1          IPC         172580920.677083         5.104167    src/ipc/handler.c:513 	ipc: comp 1 trigger 0x50000 failed -22

```

```
[ 7295.580543] systemd[1]: Starting Network Name Resolution...
[ 7295.583726] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[ 7295.583747] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 7295.599397] systemd[1]: Started Network Name Resolution.
[ 7295.661268] PM: suspend exit
[ 7295.695959] usb 1-1: new high-speed USB device number 9 using xhci_hcd
[ 7295.823226] usb 1-1: New USB device found, idVendor=05e3, idProduct=0608, bcdDevice=85.36
[ 7295.823227] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 7295.823228] usb 1-1: Product: USB2.0 Hub
[ 7295.824481] hub 1-1:1.0: USB hub found
[ 7295.824755] hub 1-1:1.0: 4 ports detected
[ 7295.885952] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60040000 size 12
[ 7295.885962] sof-audio-pci 0000:00:1f.3: unknown ROM status value 0000001c
[ 7295.886292] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000086 trace 0x00000086
[ 7295.886303] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[ 7295.886305]  HDA Analog: ASoC: trigger FE failed -110
[ 7295.886309] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[ 7295.886318] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 7295.886471] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[ 7295.886472]  HDA Analog: ASoC: trigger FE failed -22
[ 7296.083419] systemd[1]: systemd-journald.service: Main process exited, code=dumped, status=6/ABRT
[ 7296.083455] systemd[1]: systemd-journald.service: Failed with result 'watchdog'.
[ 7296.083791] systemd[1]: systemd-journald.service: Service has no hold-off time, scheduling restart.
[ 7296.102908] usb 1-1.1: new high-speed USB device number 10 using xhci_hcd
[ 7296.111042] systemd-journald[19336]: File /var/log/journal/c40c465ac40b49188d6908d8582430d7/system.journal corrupted or uncleanly shut down, renaming and replacing.
[ 7296.147884] systemd-journald[19336]: File /var/log/journal/c40c465ac40b49188d6908d8582430d7/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[ 7296.196164] usb 1-1.1: New USB device found, idVendor=0b95, idProduct=7720, bcdDevice= 0.01
[ 7296.196166] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7296.196167] usb 1-1.1: Product: AX88772A
[ 7296.196169] usb 1-1.1: Manufacturer: ASIX Elec. Corp.
[ 7296.196169] usb 1-1.1: SerialNumber: 0004C4
[ 7296.494534] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[ 7296.494543] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 7296.614743] asix 1-1.1:1.0 eth0: register 'asix' at usb-0000:00:14.0-1.1, ASIX AX88772 USB 2.0 Ethernet, 00:0e:c6:b7:2c:8c
[ 7296.680219] usb 1-1.4: new low-speed USB device number 11 using xhci_hcd
[ 7296.761109] usb 1-1.4: New USB device found, idVendor=0461, idProduct=4d15, bcdDevice= 2.00
[ 7296.761114] usb 1-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 7296.761117] usb 1-1.4: Product: USB Optical Mouse
[ 7296.767356] input: USB Optical Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.4/1-1.4:1.0/0003:0461:4D15.0003/input/input11
[ 7296.767610] hid-generic 0003:0461:4D15.0003: input,hidraw0: USB HID v1.11 Mouse [USB Optical Mouse] on usb-0000:00:14.0-1.4/input0
[ 7296.783251] asix 1-1.1:1.0 enx000ec6b72c8c: renamed from eth0
[ 7296.797938] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[ 7296.797955] sof-audio-pci 0000:00:1f.3: unknown ROM status value 0000001c
[ 7296.798338] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000087 trace 0x00000087
[ 7296.798355] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 7296.798384] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[... cut ...] pc timed out for 0x80010000 size 120
[ 7297.102218] sof-audio-pci 0000:00:1f.3: unknown ROM status value 0000001c
[ 7297.102601] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000087 trace 0x00000087
[ 7297.102616] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 7297.102621] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[ 7297.102726] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0

```

keyonjie, 2019-05-24:
> [ 7297.102726] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
> ```

@ClarexZhou can you paste the log of 13th iteration also? I doubt error happened earlier from the log you pasted(14th one).

ClarexZhou, 2019-05-24:
@keyonjie  previous logs are already cleaned. will past once reproduced again.

And ipc error for 0x60040000 size 12 is occurred during S3 with playback test.  Also add previous logs.
[s3_test_12_dmesg.log](https://github.com/thesofproject/sof/files/3215169/s3_test_12_dmesg.log)
[s3_test_12_logger_error.log](https://github.com/thesofproject/sof/files/3215170/s3_test_12_logger_error.log)
[s3_test_13_dmesg.log](https://github.com/thesofproject/sof/files/3215171/s3_test_13_dmesg.log)
[s3_test_13_logger_error.log](https://github.com/thesofproject/sof/files/3215172/s3_test_13_logger_error.log)

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         COMP          14412256.614583  14412257.000000 rc/audio/component.c:140 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_START
    0      1          IPC          14412261.145833         4.531250 src/audio/pipeline.c:606 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 1
    0      1          IPC          14412275.416667        14.270833    src/ipc/handler.c:513 	ipc: comp 1 trigger 0x40000 failed -22
    0      1         COMP          14412511.354167       235.937500 rc/audio/component.c:161 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC          14412515.468750         4.114583 src/audio/pipeline.c:606 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 0
    0      1          IPC          14412520.572917         5.104167    src/ipc/handler.c:513 	ipc: comp 1 trigger 0x50000 failed -22
    0      1          DMA          15170268.281250    757747.687500 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1

```

```
[  269.140855] PM: suspend entry (s2idle)
[  269.140858] PM: Syncing filesystems ... done.
[  269.142088] Freezing user space processes ... (elapsed 0.002 seconds) done.
[  269.144551] OOM killer disabled.
[  269.144552] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  269.145900] printk: Suspending console(s) (use no_console_suspend to debug)
[  269.249382] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 5
[  269.249414] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  269.249563] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  269.249572] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  269.249808] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  269.249811] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=5
[  269.249814] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  269.249988] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  270.267569] PM: Some devices failed to suspend, or early wake event detected
[  270.267639] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[  270.267696] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[  270.267710] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[  270.470813] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[  272.330716] OOM killer enabled.
[  272.330716] Restarting tasks ... done.
[  272.338738] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  272.338753] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  272.338926] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60040000 size 12
[  272.338948]  HDA Analog: ASoC: trigger FE failed -22
[  272.338949] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  272.338957] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  272.339109] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[  272.339111]  HDA Analog: ASoC: trigger FE failed -22
[  272.403217] PM: suspend exit
[  272.502221] asix 1-1.1:1.0 enx000ec6b72c8c: link up, 100Mbps, full-duplex, lpa 0xCDE1
[  272.945156] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  272.945166] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  272.945383] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  272.945416] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  272.945629] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  272.945756] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0

```

keyonjie, 2019-05-24:
@ClarexZhou yes, better to paste logs for iteration:
[n-1]: normal iteration(no any errors in dmesg and sof-logger) right before the the error happen, and
[n]: the first error iteration.

BTW, better to apply https://github.com/thesofproject/linux/pull/888 to survive the sof-logger error logs.

ClarexZhou, 2019-05-27:
Track ipc error in #1477

ClarexZhou, 2019-05-28:
Error occurred during s3 with playback stress test at 29th.
***Environment***
Kernel: https://github.com/lyakh/linux/tree/ipc-timeout commit 9453da1 (with PR895)
FW:https://github.com/lyakh/sof/tree/ipc-timeout commit 47dc14d (with PR1354 and PR1470)
Kconfig: disable CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE in kernel config.

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          EDF         14585734516.197918              nan c/lib/edf_schedule.c:359 	unexpected task state 5 at edf_task completion
    0      1         COMP         14676669603.281250  90935088.000000 rc/audio/component.c:140 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_START
    0      1          IPC         14676669609.635418         6.354167 src/audio/pipeline.c:606 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 1
    0      1          IPC         14676669616.614584         6.979167    src/ipc/handler.c:513 	ipc: comp 1 trigger 0x40000 failed -22
    0      1         COMP         14677056627.500000    387010.875000 rc/audio/component.c:161 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC         14677056633.489584         5.989583 src/audio/pipeline.c:606 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 0
    0      1          IPC         14677056640.364584         6.875000    src/ipc/handler.c:513 	ipc: comp 1 trigger 0x50000 failed -22

```
```
[  421.876128] PM: suspend entry (s2idle)
[  421.876132] PM: Syncing filesystems ... done.
[  421.879257] Freezing user space processes ... 
[  424.355763] Freezing of tasks aborted after 2.476 seconds (1 tasks refusing to freeze, wq_busy=0):
[  424.355772] OOM killer enabled.
[  424.355774] Restarting tasks ... done.
[  424.410778] PM: suspend exit
[  425.805657] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  425.805678] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  426.113457] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60040000 size 12
[  426.113475] sof-audio-pci 0000:00:1f.3: unknown ROM status value 00000027
[  426.113717] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000087 trace 0x00000087
[  426.113727] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[  426.113732] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  426.113886]  HDA Analog: ASoC: trigger FE failed -110
[  426.113889] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.113896] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.113900] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.113904] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.113907] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.113912] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.113914] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.113943] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  426.114103] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  426.114141] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  426.114167] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114407] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114417] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114427] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114450] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114454] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114481] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114558] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114767] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114796] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114802] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114809] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114823] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114854] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.114900] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.115202] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.115242] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.115292] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.115302] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  426.115316] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.115427] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[  426.115438]  HDA Analog: ASoC: trigger FE failed -22
[  426.115561] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  426.115592] sof-audio-pci 0000:00:1f.3: error: trace IO error

```

[s3_test_28_logger_all.log](https://github.com/thesofproject/sof/files/3225913/s3_test_28_logger_all.log)
[s3_test_28_logger_error.log](https://github.com/thesofproject/sof/files/3225914/s3_test_28_logger_error.log)
[s3_test_29_logger_all.log](https://github.com/thesofproject/sof/files/3225916/s3_test_29_logger_all.log)
[s3_test_29_logger_error.log](https://github.com/thesofproject/sof/files/3225917/s3_test_29_logger_error.log)
[test_28.log](https://github.com/thesofproject/sof/files/3225918/test_28.log)
[test_29.log](https://github.com/thesofproject/sof/files/3225919/test_29.log)

lgirdwood, 2019-05-28:
@ClarexZhou your last comment is a different bug, can you open a new issue for it. Your last comment looks like a FW crash as I can see a scheduler error message which puts pipeline into an inconsistent state..
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          EDF         14585734516.197918              nan c/lib/edf_schedule.c:359 	unexpected task state 5 at edf_task completion
    0      1         COMP         14676669603.281250  90935088.000000 rc/audio/component.c:140 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_START
    0      1          IPC         14676669609.635418         6.354167 src/audio/pipeline.c:606 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 1
    0      1          IPC         14676669616.614584         6.979167    src/ipc/handler.c:513 	ipc: comp 1 trigger 0x40000 failed -22
    0      1         COMP         14677056627.500000    387010.875000 rc/audio/component.c:161 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC         14677056633.489584         5.989583 src/audio/pipeline.c:606 	pipeline_trigger() error: ret = -22, host->comp.id = 1, cmd = 0
    0      1          IPC         14677056640.364584         6.875000    src/ipc/handler.c:513 	ipc: comp 1 trigger 0x50000 failed -22
```

ClarexZhou, 2019-05-28:
keep monitor, and will report new issue once reproduce the scheduler error message caused ipc timed out issue again.

ClarexZhou, 2019-05-28:
Error occurred during aplay stress test at 25th.

**Environment**
Kernel: https://github.com/lyakh/linux/tree/ipc-timeout commit 9453da1 (with PR895)
FW:https://github.com/lyakh/sof/tree/ipc-timeout commit 47dc14d (with PR1354 and PR1470)
Kconfig: disable CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE in kernel config.
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         COMP         111679722066.145844              nan rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         111679722113.125000        46.979168 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         111679722118.593750         5.468750 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     111679722125.416672         6.822917 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1         COMP         111679725814.635422      3689.218750 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         111679725860.781250        46.145832 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         111679725866.093750         5.312500 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     111679725872.760422         6.666667 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1         COMP         111679729564.531250      3691.770752 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         111679729610.729172        46.197918 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         111679729616.197922         5.468750 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     111679729622.916672         6.718750 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1         COMP         111679733314.427094      3691.510498 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         111679733360.625000        46.197918 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         111679733366.197922         5.572917 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     111679733372.760422         6.562500 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1         COMP         111679737064.687500      3691.927002 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         111679737110.885422        46.197918 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         111679737116.458344         5.572917 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     111679737123.020844         6.562500 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1         COMP         111679740814.479172      3691.458252 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         111679740860.989594        46.510418 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         111679740866.302094         5.312500 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     111679740872.812500         6.510417 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1         COMP         111679744564.739594      3691.927002 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         111679744610.885422        46.145832 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         111679744616.197922         5.312500 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     111679744623.072922         6.875000 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1         COMP         111679748314.531250      3691.458252 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
    0      1       VOLUME         111679748360.781250        46.250000 udio/volume/volume.c:510 	volume_copy(): Failed comp_get_copy_limits()
    0      1         PIPE         111679748366.093750         5.312500 src/audio/pipeline.c:732 	pipeline_copy() error: ret = -5, start->comp.id = 2, dir = 1
    0      1         PIPE 1.6     111679748372.916672         6.822917 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1         COMP         111679752064.531250      3691.614502 rc/audio/component.c:261 	comp_get_copy_limits() error: sink component buffer has not enough free bytes for copy
[... cut ...]
[ 2459.754509] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[ 2459.754688] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0

```

[aplay_error_25_dmesg.log](https://github.com/thesofproject/sof/files/3227163/aplay_error_25_dmesg.log)
[aplay_error_25_logger_all.log](https://github.com/thesofproject/sof/files/3227164/aplay_error_25_logger_all.log)
[aplay_error_25_logger_error.log](https://github.com/thesofproject/sof/files/3227165/aplay_error_25_logger_error.log)

emilchudzik, 2019-05-31:
@ClarexZhou reporting issue on non-official branch (official is master or alpha/beta/pv branches) is only debug, so please report it directly to owner od code.

Please double check issue on master or cml-hda... branch.

jajanusz, 2019-06-17:
@emilchudzik @ClarexZhou Can we close it? Looks like it turned out to be kernel-side issue.

ClarexZhou, 2019-06-18:
Close for not reproduced

closed 2019-06-18 by ClarexZhou