thesofproject/sof #1401: [BUG] APL GCC toolchain uses older version from Cadence

tlauda, 2019-05-09:
**Describe the bug**
APL GCC toolchain uses older version from Cadence, which causes compilation failure with the latest updated headers.

lgirdwood, 2019-05-09:
@tlauda I've updated locally but I still see the same issues. Can you confirm the code is generated for CNL ?

tlauda, 2019-05-10:
@lgirdwood I haven't checked the binary, but compared GCC build logs in CI and they are basically the same, meaning same parameters are used. Also I've checked make files and other stuff and everything is the same for both APL and other platforms. I don't see any other root cause than toolchain.

lgirdwood, 2019-05-11:
@tlauda thanks, this one is confusing. I need to check again, I rebuilt APL GCC with the updates but I've lost all this work as my NUC has died. I'll revisit when I'm back from travelling.

jajanusz, 2019-06-17:
@lgirdwood @tlauda ping

lgirdwood, 2019-06-18:
@jajanusz this got stuck in my NUC failure. Will fix for 1.4

closed 2019-07-26 by lgirdwood