thesofproject/sof #1403: [BUG] Userspace structures in uapi/user are not using ABI header

lgirdwood, 2019-05-09:
**Describe the bug**
Userspace has no way of informing FW or driver of ABI used for binary kcontrol data.

**Expected behavior**
Userspace will use struct sof_abi_hdr as header for all binary kcontrol structures.

**Impact**
showstopper. will cause random breakage in the future.

Needs added to FW headers, kernel headers and userspace tools.

lgirdwood, 2019-05-09:
@plbossart fyi, needs to be done prior to next kernel release.

lgirdwood, 2019-05-09:
@plbossart I see this has just been removed from kernel uapi, so it just a FW/tool change now.

wenqingfu, 2019-05-13:
Liam agreed to fix it

lgirdwood, 2019-05-31:
@singalsu I'm looking for users of the structures in include/user/eq.h but cant find any non firmware users ? How is the ABI currently specified for this data ?

lgirdwood, 2019-05-31:
@singalsu one more thing, I also cant see where eqctl inserts ABI information, I guess this in the .txt files ?

ranj063, 2019-05-31:
@lgirdwood yes the ABI is manually added into the txt files

lgirdwood, 2019-06-05:
Merged #1517

closed 2019-06-05 by lgirdwood

singalsu, 2019-06-05:
@lgirdwood The EQ tool gets the ABI via function SOF/tools/tune/eq/eq_get_abi.m. Currently the version is hard coded there manually to 3.0.0.

I've been wondering if tools like sof-ctl could return the ABI header in usable form for Matlab code. The C headers can't be used for other languages easily.

lgirdwood, 2019-06-05:
@singalsu I intend to add some features requests for this.