thesofproject/sof #1411: [BUG] KWD: HDMI playback failure when using kwd tplg

keqiaozhang, 2019-05-13:
**Describe the bug**
HDMI playback failed with invalid channel -1, this issue only happens with sof-glk-da7219-kwd.tplg, no such issue after switching to sof-glk-da7219.tplg.

**To Reproduce**
aplay -Dhw:0,5 -f dat -c 2 audio.wav

**Expected behavior**
HDMI playback should works well without errors.

**Impact**
unable to play audio through HDMI.

**Environment**
1) kernel: sof-dev/35d7759478
2) firmware: master/3f1b92 + PR #1365
3) tplg: sof-glk-da7219-kwd.tplg.
4) platform: GLK chromebook
5) Reproducibility Rate. 100%.

**Screenshots or console output**
$ aplay -Dhw:0,5 -f dat -c 2 audio.wav  -i
Playing WAVE 'audio.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1403: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 341000
BUFFER_SIZE: 16368
BUFFER_BYTES: 65472
TICK_TIME: 0

**dmesg**
```
[  833.916271] sof-audio-pci 0000:00:0e.0: stream_tag 1
[  833.916275] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  833.916455] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[  833.916459] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 1
[  833.916462] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -22
[  833.916464]  HDMI1: ASoC: hw_params FE failed -22
[  833.916481] sof-audio-pci 0000:00:0e.0: pcm: free stream 5 dir 0
[  833.916483] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  833.916665] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  833.916679] WARNING: CPU: 2 PID: 2842 at kernel/workqueue.c:3030 __flush_work+0x19f/0x1b0
[  833.916679] Modules linked in: snd_soc_sst_bxt_da7219_max98357a asix usbnet snd_soc_hdac_hdmi snd_soc_dmic x86_pkg_temp_thermal sof_pci_dev snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda snd_soc_acpi_intel_match snd_sof iwlmvm snd_sof_xtensa_dsp snd_soc_acpi snd_hda_ext_core snd_hda_codec snd_soc_da7219 snd_hwdep snd_soc_max98357a snd_soc_core snd_hda_core snd_pcm iwlwifi efivarfs mmc_block sdhci_pci cqhci sdhci intel_lpss_pci xhci_pci intel_lpss mfd_core xhci_hcd
[  833.916700] CPU: 2 PID: 2842 Comm: aplay Tainted: G        W         5.1.0+ #1
[  833.916701] Hardware name: Google Bobba/Bobba, BIOS Google_Bobba.11825.0.2019_03_06_2015 03/06/2019
[  833.916704] RIP: 0010:__flush_work+0x19f/0x1b0
[  833.916706] Code: ff 8b 4d 00 48 8b 55 08 83 e1 08 48 0f ba 6d 00 03 80 c9 f0 eb c6 41 c6 45 00 00 fb 31 c0 e9 e1 fe ff ff 0f 0b e9 da fe ff ff <0f> 0b 31 c0 e9 d1 fe ff ff e8 33 2d fe ff 0f 1f 00 31 f6 e9 49 fe
[  833.916707] RSP: 0018:ffff915481f6fbd0 EFLAGS: 00010246
[  833.916708] RAX: 0000000000000000 RBX: ffff8b2bb9285428 RCX: 0000000000000000
[  833.916709] RDX: 0000000000000001 RSI: 0000000000000001 RDI: ffff8b2bb9285428
[  833.916710] RBP: ffff8b2bb9285428 R08: 000000000000120c R09: 0000000000000004
[  833.916710] R10: ffff9154812d3bb0 R11: 0000000000000001 R12: 0000000000000000
[  833.916711] R13: ffff915481f6fc68 R14: ffffffff93e7c890 R15: 0000000000000000
[  833.916712] FS:  00007f7fce2d4740(0000) GS:ffff8b2bbbb00000(0000) knlGS:0000000000000000
[  833.916713] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  833.916714] CR2: 00007f7fce98e000 CR3: 0000000147ec0000 CR4: 0000000000340ee0
[  833.916715] Call Trace:
[  833.916725]  ? ipc_log_header+0x1a6/0x4b0 [snd_sof]
[  833.916728]  ? sof_ipc_tx_message_unlocked+0x12d/0x2d0 [snd_sof]
[  833.916730]  __cancel_work_timer+0x11f/0x1a0
[  833.916733]  ? sof_ipc_tx_message+0x61/0x80 [snd_sof]
[  833.916735]  sof_pcm_hw_free+0x158/0x170 [snd_sof]
[  833.916742]  soc_pcm_components_hw_free+0x5c/0x80 [snd_soc_core]
[  833.916747]  soc_pcm_hw_free+0x121/0x1d0 [snd_soc_core]
[  833.916752]  dpcm_fe_dai_hw_free+0x73/0x100 [snd_soc_core]
[  833.916759]  snd_pcm_hw_params+0x12d/0x5d0 [snd_pcm]
[  833.916764]  ? _cond_resched+0x10/0x40
[  833.916766]  ? __kmalloc_track_caller+0x130/0x1b0
[  833.916770]  snd_pcm_common_ioctl+0x19e/0xc80 [snd_pcm]
[  833.916773]  snd_pcm_ioctl+0x25/0x30 [snd_pcm]
[  833.916775]  do_vfs_ioctl+0x9f/0x620
[  833.916778]  ksys_ioctl+0x6b/0x80
[  833.916780]  __x64_sys_ioctl+0x11/0x20
[  833.916782]  do_syscall_64+0x43/0xf0
[  833.916785]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  833.916787] RIP: 0033:0x7f7fce57c3c7
[  833.916789] Code: 00 00 90 48 8b 05 c9 3a 0d 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 99 3a 0d 00 f7 d8 64 89 01 48
[  833.916790] RSP: 002b:00007fff006a8768 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  833.916791] RAX: ffffffffffffffda RBX: 000055a6db4c6b70 RCX: 00007f7fce57c3c7
[  833.916792] RDX: 00007fff006a8900 RSI: 00000000c2604111 RDI: 0000000000000004
[  833.916792] RBP: 00007fff006a8900 R08: 0000000000000000 R09: 0000000000000000
[  833.916793] R10: 0000000000000004 R11: 0000000000000246 R12: 000055a6db4c6bf0
[  833.916794] R13: 00007fff006a8794 R14: 000000000000bb80 R15: 000000000000bb80
[  833.916796] ---[ end trace 1868630c38cc40f0 ]---
```
**logger**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          MEM             51937.968750     51937.968750      src/lib/alloc.c:271    error: 342 blocks needed for allocation but only 341 blocks are remaining
    0      1          DMA             65301.510417     13363.541992 intel/cavs/hda-dma.c:698    hda-dmac: 7 channel 0 HDA Link DMA doesn't support irq scheduling
    0      1         PIPE             65308.750000         7.239583 src/audio/pipeline.c:359    pipeline_prepare() error: ret = -22,dev->comp.id = 29
    0      1          IPC             65315.000000         6.250000    src/ipc/handler.c:326    ipc: pipe 5 comp 29 prepare failed -22
    0      1          DMA             65454.687500       139.687500 intel/cavs/hda-dma.c:616    hda-dmac: 5 invalid channel -1
```
**logger -t**
```
    0      2          DAI 5.33        65293.541667         7.291667      src/audio/dai.c:445    dai_prepare()
    0      2          DMA             65297.864583         4.322917 intel/cavs/hda-dma.c:685    hda-dmac: 7 channel 0 -> config
    0      1          DMA             65301.510417         3.645833 intel/cavs/hda-dma.c:698    hda-dmac: 7 channel 0 HDA Link DMA doesn't support irq scheduling
    0      1         PIPE             65308.750000         7.239583 src/audio/pipeline.c:359    pipeline_prepare() error: ret = -22,dev->comp.id = 29
    0      1          IPC             65315.000000         6.250000    src/ipc/handler.c:326    ipc: pipe 5 comp 29 prepare failed -22
    0      2         PIPE 5.34        65321.354167         6.354167 src/audio/pipeline.c:634    pipeline_reset()
    0      2         HOST             65325.468750         4.114583     src/audio/host.c:699    host_reset()
```

[dmesg.log](https://github.com/thesofproject/sof/files/3171620/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3171621/logger.log)
[logger-t.log](https://github.com/thesofproject/sof/files/3171622/logger-t.log)

keyonjie, 2019-05-14:
hda-dmac: 7 channel 0 HDA Link DMA doesn't support irq scheduling

@tlauda @mrajwa @jajanusz  what happens here?

jajanusz, 2019-05-14:
@keqiaozhang @keyonjie Did #1430 help?

keqiaozhang, 2019-05-15:
@jajanusz 
Confirmed that this issue can be fixed by PR #1430.

Test recipe:
Firmware: glk-012-stable-branch/59ad3 + PR #1430 
Kernel: sof-dev/35d7759478
Platform: GLK Ubuntu 18.04

keqiaozhang, 2019-05-16:
This issue is fixed on glk-012-stable-branch/94e0f0b.

closed 2019-05-16 by keqiaozhang