thesofproject/sof #1414: [BUG] WoV:  DSP panic after dmic(pcm8) recording is waked up

keqiaozhang, 2019-05-13:
**Describe the bug**
This issue happens after waking up the dmic recording via the "keyword"(Clap hard), then dmic recording will stop there(need ctl+c to kill it), the dmesg shows that DSP is crashed. 
I can only reproduce this issue with test recipe 1 and cannot reproduce this issue with test recipe 2.

**To Reproduce**
1. arecord -Dhw:0,8 -M -N -r 16000 -f s16_le -c 2 wov.wav -vvv
2. clap hard to wake up the recording
3. you will see dmic recording wake up and stop there.

**Expected behavior**
no panic after waking up..

**Impact**
DSP crash and firmware oops.

**test recipe 1**
kernel: cros-sof-v4.14-rebase /dc349dd
firmware: master/3f1b92 + PR #1365
tplg: sof-glk-da7219-kwd.tplg.
platform: GLK chromebook chrome OS R73-11559.0.0
Reproducibility Rate. 100%.

**test recipe 2**
kernel: sof-dev /35d775
firmware: master/3f1b92 + PR #1365
tplg: sof-glk-da7219-kwd.tplg.
platform: GLK ubuntu 18.04
Reproducibility Rate. 0%.

**Screenshots or console output**
```localhost ~ # arecord -Dhw:0,8 -M -N -r 16000 -f s16_le -c 2 wov.wav -vvv
Recording WAVE 'wov.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
Hardware PCM card 0 'sof-glkda7219max' device 8 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 16
  buffer_size  : 8016
  period_size  : 2016
  period_time  : 126000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2016
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8016
  silence_threshold: 0
  silence_size : 0
  boundary     : 9025213653250473984
  appl_ptr     : 0
  hw_ptr       : 0
mmap_area[0] = 0x78215f6a8000,0,32 (16)
mmap_area[1] = 0x78215f6a8000,16,32 (16)
Max peak (4032 samples): 0x00000084 #                    0%
Max peak (4032 samples): 0x0000005a #                    0%
overrun!!! (at least 0.041 ms long)
Status:
  state       : XRUN
  trigger_time: 87.121772
  tstamp      : 0.000000
  delay       : 0
  avail       : 9728
  avail_max   : 9728
Max peak (4032 samples): 0x0000055e #                    4%
[... 186 lines cut ...]
    0      2          DAI 8.54      6738496.770833         3.281250      src/audio/dai.c:572    dai_comp_trigger(), PAUSE/STOP
    0      2          DMA           6738500.364583         3.593750 src/drivers/dw/dma.c:460    dw_dma_stop(): dma 0 channel 4 stop
    0      2         DMIC           6738508.906250         8.541667 rs/intel/cavs/dmic.c:1448   dmic_trigger()
    0      2         DMIC           6738512.656250         3.750000 rs/intel/cavs/dmic.c:1380   dmic_stop()
    0      2         DMIC           6738516.822917         4.166667 rs/intel/cavs/dmic.c:1400   dmic_stop(), dmic_active_fifos = 1
    0      1         PIPE           6738525.572917         8.750000 src/audio/pipeline.c:732    pipeline_copy() error: ret = -5, start->comp.id = 54, dir = 0
    0      1         PIPE 8.55      6738532.500000         6.927083 src/audio/pipeline.c:843    pipeline_xrun_recover()
    0      2         PIPE 8.55      6738537.447917         4.947917 src/audio/pipeline.c:352    pipeline_prepare()
    0      2          DAI 8.54      6738541.562500         4.114583      src/audio/dai.c:445    dai_prepare()
    0      2         PIPE 8.55      6738546.458333         4.895833 src/audio/pipeline.c:579    pipeline_trigger()
    0      2          DAI 8.54      6738550.208333         3.750000      src/audio/dai.c:508    dai_comp_trigger(), command = 1
    0      2          DAI 8.54      6738553.750000         3.541667      src/audio/dai.c:519    dai_comp_trigger(), START
    0      2         DMIC           6738564.531250        10.781250 rs/intel/cavs/dmic.c:1448   dmic_trigger()
    0      2         DMIC           6738568.593750         4.062500 rs/intel/cavs/dmic.c:1280   dmic_start()
    0      2         DMIC           6738572.083333         3.489583 rs/intel/cavs/dmic.c:1296   dmic_start(), dmic->fifo_b
    0      2         DMIC           6738576.041667         3.958333 rs/intel/cavs/dmic.c:1323   dmic_start(), mic_a = 1, mic_b = 1, fir_a = 1, fir_b = 1
    0      2         DMIC           6738580.572917         4.531250 rs/intel/cavs/dmic.c:1323   dmic_start(), mic_a = 0, mic_b = 0, fir_a = 0, fir_b = 0
    0      2         DMIC           6738585.260417         4.687500 rs/intel/cavs/dmic.c:1359   dmic_start(), dmic_active_fifos = 1
    0      2         DMIC           6738590.000000         4.739583 rs/intel/cavs/dmic.c:1371   dmic_start(), done
    0      2           SA           6750178.958333     11588.958008      src/lib/agent.c:74     validate(), idle longer than timeout, delta = 19065543
```

[dmesg.log](https://github.com/thesofproject/sof/files/3172046/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3172048/logger.log)
[logger-t.log](https://github.com/thesofproject/sof/files/3172050/logger-t.log)

keyonjie, 2019-05-14:
Tomek's sof PR#1415 should be helpful for this?

keqiaozhang, 2019-05-14:
No DSP panic with below recipe:
firmware: sof-master/b73b9a + applying PR #1383 + reverting PR #1363 
kernel: cros-sof-v4.14-rebase /dc349dd

keqiaozhang, 2019-05-14:
Sorry for my mistake.
I found this issue with a workaround patch, but this workaround has some conflicts with PR #1363. but I didn't release it. So after removing the workaround and keep PR #1363 , there's also no DSP panic.  The results of sof-dev and cros-sof-v4.14-rebase are same now. 
So let's close this issue.
The workaround is from Keyon, I pasted it as below for your reference.
```
diff --git a/src/audio/detect_test.c b/src/audio/detect_test.c
index 8fb1a5536177..fda4c4a2909f 100644
--- a/src/audio/detect_test.c
+++ b/src/audio/detect_test.c
@@ -299,6 +299,8 @@ static int test_keyword_set_config(struct comp_dev *dev,
        struct sof_detect_test_config *cfg;
        size_t bs;

+       return 0;
+
        /* Copy new config, find size from header */
        cfg = (struct sof_detect_test_config *)cdata->data->data;
        bs = cfg->size; 
```

closed 2019-05-14 by keqiaozhang