thesofproject/sof #1421: [BUG][CML]Aplay not exist after do Suspeand & Resume with playback_Analog_50 times

YvonneYang2, 2019-05-14:
**Describe the bug**
Aplay not exist after do Suspeand & Resume with playback_Analog_50 times

**To Reproduce**

1. Set Suspeand & Resume with playback_Analog_50 times via command “aplay -Dhw:0,0 -r48000 -c2 -fs16_le /dev/zero & sudo ./stress-s3.sh 50 1 1”

**Expected behavior**
Streaming will continuous playback during the 50 times test

**Impact**
Show "aplay not exist", but do aplay again can output audio.

**Environment**
Rander UCM https://github.com/RanderWang/intel-UCM   
Kernel repo：9494607 https://github.com/thesofproject/linux/tree/release/sof-v5.0 
Firmware repo: b73b9a6 https://github.com/thesofproject/sof/commits/master +PR1354  
Topology: https://github.com/thesofproject/sof/commits/master . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg
platform: CML laptop with ALC3204 codec

**dmesg**
```
[  285.490582] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  285.490595] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  285.490767] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  285.490770] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  286.445762] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  286.445804] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  291.019550] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  291.019577] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  291.019833] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  291.019845] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[  291.019934] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  291.019939] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  291.020147] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  291.020171] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  291.020496] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  291.020633] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  291.945521] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  291.945553] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  293.134455] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  293.134800] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  293.134840] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  293.134843] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[  293.134844] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  293.134869] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
```

[AplayautoDisappeared_dmesg.log](https://github.com/thesofproject/sof/files/3176098/AplayautoDisappeared_dmesg.log)
[Aplayauto.log](https://github.com/thesofproject/sof/files/3176099/Aplayauto.log)

tlauda, 2019-05-15:
@YvonneYang2 Looks like kernel issue:
```
[  252.513350] sof-audio-pci 0000:00:1f.3: hda: prepare stream dir 0
[  252.513354] sof-audio-pci 0000:00:1f.3: ASoC: cpu DAI prepare error: -16
[  252.513357]  Analog Playback and Capture: ASoC: backend prepare failed -16
```

Please close this one and reopen in thesofproject/linux.

YvonneYang2, 2019-05-16:
OK. Track the issue with #951. Close the bug.

closed 2019-05-16 by YvonneYang2