thesofproject/sof #1422: [BUG][CML]ipc timed out for 0x80010000: GLB_DAI_MSG: CONFIG after system boot up.

ClarexZhou, 2019-05-14:
**Describe the bug**
sudo reboot to do system reboot, ipc timed out for 0x80010000: GLB_DAI_MSG: CONFIG after system boot up.

**To Reproduce**
sudo reboot to do system reboot 

**Expected behavior**
Bootup without any issues.

**Impact**
IPC timed out.

**Environment**
platform: CML laptop with ALC3204 codec
Rander UCM https://github.com/RanderWang/intel-UCM   
Kernel repo：9494607 https://github.com/thesofproject/linux/tree/release/sof-v5.0 
Firmware repo: b73b9a6 https://github.com/thesofproject/sof/commits/master +PR1354  
Topology: https://github.com/thesofproject/sof/commits/master . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

Error sof-logger:
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA           4499110.468750   4499110.500000 intel/cavs/hda-dma.c:616 	hda-dmac: 5 invalid channel -1
```
Dmesg:
```
[    3.370305] usb 1-4: Manufacturer: VIA Technologies Inc.         
[    3.370306] usb 1-4: SerialNumber: 0000000000000001
[    3.411256] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[    3.411279] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[    3.662275] usb 1-6: new high-speed USB device number 5 using xhci_hcd
[    3.747224] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[    3.761690] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[    3.812804] usb 1-6: New USB device found, idVendor=0bda, idProduct=5538, bcdDevice=88.07
[    3.812810] usb 1-6: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[    3.812814] usb 1-6: Product: Integrated_Webcam_HD
[    3.812818] usb 1-6: Manufacturer: CKFHH66O4634000251A0
[    3.812820] usb 1-6: SerialNumber: 200901010001
[    3.929218] usb 1-10: new full-speed USB device number 6 using xhci_hcd
[    4.057287] usb 1-10: New USB device found, idVendor=8087, idProduct=0aaa, bcdDevice= 0.02
[    4.057295] usb 1-10: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    4.445659] ax88179_178a 1-3:1.0 eth0: register 'ax88179_178a' at usb-0000:00:14.0-3, ASIX AX88179 USB 3.0 Gigabit Ethernet, 00:0e:c6:a1:e5:69
[    4.445755] usbcore: registered new interface driver ax88179_178a
[    4.453208] ax88179_178a 1-3:1.0 enx000ec6a1e569: renamed from eth0
[    4.798306] systemd-udevd (1694) used greatest stack depth: 11912 bytes left
[    5.602555] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[    5.602561] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[    5.602564] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[    5.602566] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[    5.602840] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    5.903485] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[    5.903508] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    5.903737] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[    5.903747] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[    5.903751] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[    5.903757] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[    5.903762] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110
[    5.903777]  Analog Playback and Capture: ASoC: hw_params BE failed -110
[    5.903781]  HDA Analog: ASoC: hw_params BE failed -110
[    5.903794] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[    5.903800] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[    6.207484] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[    6.207510] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    6.207806] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[    6.207819] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[    6.207823] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[    6.208005] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    6.511317] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[    6.511343] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[... 54 lines cut ...]
[    8.335875] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[    8.335878] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[    8.335883] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[    8.335888] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110
[    8.335899]  Analog Playback and Capture: ASoC: hw_params BE failed -110
[    8.335903]  HDA Analog: ASoC: hw_params BE failed -110
[    8.335914] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[    8.335919] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[    8.639551] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[    8.639576] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    8.639871] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[    8.639883] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[    8.639886] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[    8.640070] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[    8.640481] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[    8.640486] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[    8.640488] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[    8.640490] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[    8.640687] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    8.943323] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[    8.943348] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    8.943643] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace
```
[ipcTimedoutBootUp_dmesg.log](https://github.com/thesofproject/sof/files/3176420/ipcTimedoutBootUp_dmesg.log)
[ipcTimedOut_Logger.log](https://github.com/thesofproject/sof/files/3176421/ipcTimedOut_Logger.log)

ClarexZhou, 2019-05-14:
Error occurred twice (2nd time reload and the 34th reload)during modules reload test on cml laptop.
Error logger
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA          12974929.583333  12974930.000000 intel/cavs/hda-dma.c:616 	hda-dmac: 5 invalid channel -1
```

```
[  109.766051] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  109.766327] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: GET_VALUE
[  109.766336] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[  110.749197] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  110.749231] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  112.498123] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[  112.498130] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  112.498134] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  112.498138] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  112.498447] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  112.800134] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  112.800153] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  112.800384] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  112.800393] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  112.800398] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  112.800405] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[  112.800412] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110
[  112.800428]  Analog Playback and Capture: ASoC: hw_params BE failed -110
[  112.800433]  HDA Analog: ASoC: hw_params BE failed -110
[  112.800449] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  112.800458] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  113.104123] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[  113.104150] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  113.104389] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  113.104411] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  113.104417] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  113.104593] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  113.408068] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  113.408098] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  113.408406] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  113.408433] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[  113.408439] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  113.408447] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[  113.408456] sof-audio-pci 0000:00
```
[logger_all.log](https://github.com/thesofproject/sof/files/3177516/logger_all.log)
[boot_fail.log](https://github.com/thesofproject/sof/files/3177518/boot_fail.log)

tlauda, 2019-05-14:
@ClarexZhou Please retest without #1354 applied.

YvonneYang2, 2019-05-15:
Error occurred once when do reboot stress test 53 times on CFL. 

**ENV**
Rander UCM https://github.com/RanderWang/intel-UCM   
Kernel repo：9494607 https://github.com/thesofproject/linux/tree/release/sof-v5.0 
Firmware repo: b73b9a6 https://github.com/thesofproject/sof/commits/master +PR1354  
Topology: https://github.com/thesofproject/sof/commits/master . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg
platform: CFL laptop with ALC3254 codec 

**Error logger**
```
[  122.913383] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  122.913387] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  122.913391] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  122.913932] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  126.221919] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[  126.238728] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[  164.086554] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[  194.456910] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 1
[  194.456917] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  194.456922] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  194.456925] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  194.457196] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  194.774433] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  194.774448] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  194.774707] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  194.774730] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  194.774735] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  194.774741] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[  194.774748] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110
[  194.774761]  Analog Playback and Capture: ASoC: hw_params BE failed -110
[  194.774765]  HDA Analog: ASoC: hw_params BE failed -110
[  194.774777] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 1
[  194.774785] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  195.085576] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[  195.085591] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  195.085857] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  195.085881] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  195.085886] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  195.086037] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  195.395553] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  195.395568] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  195.395832] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  195.395847] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[  195.395852] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  195.395859] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[  195.395866] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110
[  195.395878]  Analog Playback and Capture: ASoC: hw_params BE failed -110
[  195.395882]  HDA Analog: ASoC: hw_params BE failed -110
[  195.395894] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 1
[  195.395901] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  195.703172] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[  195.703191] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  195.703484] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  195.703506] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[  195.703511] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  195.703645] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  196.014322] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  196.014339] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
```
[dmesg_failed_52.log](https://github.com/thesofproject/sof/files/3180593/dmesg_failed_52.log)
[logger-t_failed_52.log](https://github.com/thesofproject/sof/files/3180607/logger-t_failed_52.log)

ranj063, 2019-05-22:
@ClarexZhou @YvonneYang2 the recipe used for this issue is incorrect. You cannot use FW PR 1354 without taking kernel PR 895. Please repeat this test by either taking both or removing both.

lgirdwood, 2019-05-22:
@mengdonglin can we also retest with trace disabled here too.

ClarexZhou, 2019-05-23:
@ranj063 . previously we used are with kernel PR 895+ FW PR 1354 
Kernel repo：9494607 https://github.com/thesofproject/linux/tree/release/sof-v5.0   -> is merged #895 on May 14.
Firmware repo: b73b9a6 https://github.com/thesofproject/sof/commits/master +PR1354  

And '0x80010000: GLB_DAI_MSG: CONFIG' ipc timed out also occurred on May 22 with new PR 895+ FW PR 1354 
Kernel by Guennadi: https://drive.google.com/drive/folders/157Bo5KSp73miVnh1rFrIcEgqBqWFwaCX
Sof by Emil:  https://drive.google.com/drive/folders/1BdcrVl23tiIAbmg8Tpr4bR8vhrnyWQlU 

```
[ 3693.249907] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[ 3693.249914] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[ 3693.249918] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[ 3693.249921] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[ 3693.250668] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 3693.558958] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[ 3693.558978] sof-audio-pci 0000:00:1f.3: unknown ROM status value ffffffff
[ 3693.559301] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xffffffff trace 0xffffffff
[ 3693.559317] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[ 3693.559323] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[ 3693.559331] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110
[ 3693.559346]  Analog Playback and Capture: ASoC: hw_params BE failed -110
[ 3693.559351]  HDA Analog: ASoC: hw_params BE failed -110
[ 3693.559366] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[ 3693.559375] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 3693.862923] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[ 3693.862942] sof-audio-pci 0000:00:1f.3: unknown ROM status value ffffffff
[ 3693.863265] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xffffffff trace 0xffffffff
[ 3693.863280] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[ 3693.863904] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[ 3715.030269] sof-audio-pci 0000:00:1f.3: pcm: open stream 5 dir 0
[ 3715.030278] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[ 3715.030282] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[ 3715.030286] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[ 3715.030699] HDMI HDA Codec ehdaudio0D2: ASoC: can't set intel-hdmi-hifi3 hw params: -19
[ 3715.030713]  
```
[waspS3notEnter_ipcwhenAplay_soflogger.log](https://github.com/thesofproject/sof/files/3209938/waspS3notEnter_ipcwhenAplay_soflogger.log)

[waspS3notEnter_ipcwhenAplay_dmesg.log](https://github.com/thesofproject/sof/files/3209937/waspS3notEnter_ipcwhenAplay_dmesg.log)

ClarexZhou, 2019-05-23:
0x80010000: GLB_DAI_MSG: CONFIG ipc timed out occurred during aplay -d 1 stress test. System response very slow then ipc timed out.

***Environment***
disable CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE in kernel config.
Kernel:
https://github.com/lyakh/linux/tree/ipc-20190521
commit 5969756 ASoC: SOF: hda: couple host and link DMA during FE hw_free , PR 895
FW:
https://github.com/lyakh/sof/tree/ipc-timeout
commit: b527fd7 dai: assign/free link DMA channel during dai_config() , PR1354

[wasp_aplay_dmesg_18.log](https://github.com/thesofproject/sof/files/3211956/wasp_aplay_dmesg_18.log)
[wasp_aplay_errorlogger_18.log](https://github.com/thesofproject/sof/files/3211957/wasp_aplay_errorlogger_18.log)
[wasp_aplay_soflogger_18.log](https://github.com/thesofproject/sof/files/3211958/wasp_aplay_soflogger_18.log)

```
[  977.918121] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  977.918242] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  977.918255] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  977.918390] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  977.918441] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  978.825222] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  978.825249] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1058.761508] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[ 1058.761516] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[ 1058.761521] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[ 1058.761525] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[ 1058.762286] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1059.067585] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[ 1059.067609] sof-audio-pci 0000:00:1f.3: error: ipc fatal error - code 0000002d
[ 1059.067859] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x0000017a trace 0x0000017a
[ 1059.067876] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[ 1059.067883] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[ 1059.067893] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[ 1059.067903] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110
[ 1059.067922]  Analog Playback and Capture: ASoC: hw_params BE failed -110
[ 1059.067929]  HDA Analog: ASoC: hw_params BE failed -110
[ 1059.067950] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[ 1059.067961] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1059.371464] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[ 1059.371483] sof-audio-pci 0000:00:1f.3: error: base fw not found - code 0000002b
[ 1059.371741] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x0000017b trace 0x0000017b
[ 1059.371751] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[ 1059.371756] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[ 1059.372057] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
```
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA         17231117036.354168              nan intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231125653.177086      8616.823242 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231125921.875000       268.697906 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231126155.520836       233.645828 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231126381.614586       226.093750 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231126903.854168       522.239563 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231127067.395836       163.541672 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231127215.208336       147.812500 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231127356.458336       141.250000 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17231127504.583336       148.125000 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17232460046.250000   1332541.625000 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17232468830.937500      8784.687500 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17232469404.270836       573.333313 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17232469607.656250       203.385422 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17232469875.833336       268.177094 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         17232470565.781250       689.947937 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
[... cut ...]
    0      1          DMA         18037332714.895836       256.510406 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
    0      1          DMA         18140039371.666668 102706656.000000 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1

```
[wasp_aplay_dmesg_18.log](https://github.com/thesofproject/sof/files/3211973/wasp_aplay_dmesg_18.log)
[wasp_aplay_errorlogger_18.log](https://github.com/thesofproject/sof/files/3211974/wasp_aplay_errorlogger_18.log)
[wasp_aplay_soflogger_18.log](https://github.com/thesofproject/sof/files/3211975/wasp_aplay_soflogger_18.log)

ranj063, 2019-05-24:
@lyakh I think we have another race here. The pattern  I see in my tests with aplay stress test is that the with successive DAI_CONFIG IPC's sent, one of them at some point times out. I also think this ipc timeout might be related to the the one seen in https://github.com/thesofproject/linux/issues/958.

ranj063, 2019-05-24:
this is the patter I'm noticing with the aplay -d 1 stress test.
For all the iterations that pass, pcm open happens right after the previous aplay closes. But when the failure occurs, there's a long delay between when the previous stream closes and the next one begins. Here's an example taken from the above logs:
**[  977.918441] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  978.825222] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  978.825249] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1058.761508] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0**
[ 1058.761516] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[ 1058.761521] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[ 1058.761525] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[ 1058.762286] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1059.067585] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120

keyonjie, 2019-05-24:
@ranj063 can you paste the sof-logger log and dmesg log for the previous(the one at 978.xx second) iteration? It might already has something failed there.

ranj063, 2019-05-24:
Finally, I have a glimmer of hope here, disabling runtime PM for the HDA codec device ehdaudio0D0 allows me to run the test successfully for 1K iterations

ClarexZhou, 2019-05-24:
Error occurred at 6th S3 with playback stress test. Also add previous log

**Environment**
Kernel: 9453da1 https://github.com/lyakh/linux/tree/ipc-timeout (with PR895)
FW:b527fd7 https://github.com/lyakh/sof/tree/ipc-timeout (with PR1354)
Topology:Same as above FW repo, file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg
kconfig:
unset CONFIG_SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE
unset CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE

6th error logger
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA         145754355.677083 145754352.000000 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1
```
6th dmesg
```
[  150.988195] PM: suspend entry (s2idle)
[  150.988197] PM: Syncing filesystems ... done.
[  150.989448] Freezing user space processes ... 
[  153.875978] sof-audio-pci 0000:00:1f.3: hda: prepare stream dir 0
[  153.875996] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  154.178021] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  154.178038] sof-audio-pci 0000:00:1f.3: unknown ROM status value 00000021
[  154.178421] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000080 trace 0x00000080
[  154.178437] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  154.178443] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[  154.178448] sof-audio-pci 0000:00:1f.3: ASoC: cpu DAI prepare error: -110
[  154.178452]  Analog Playback and Capture: ASoC: backend prepare failed -110
[  154.181380] (elapsed 3.191 seconds) done.
[  154.181387] OOM killer disabled.
[  154.181388] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  154.182946] printk: Suspending console(s) (use no_console_suspend to debug)
[  192.690032] PM: Some devices failed to suspend, or early wake event detected
[  192.690106] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[  192.690144] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[  192.690158] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[  192.893501] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[  256.266939] OOM killer enabled.
[  256.266940] Restarting tasks ... 
[  256.268836] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  256.268847] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  256.269862] done.
[  256.342102] PM: suspend exit
[  256.570017] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
```

5th dmesg and no error logger
```
[  136.989536] PM: suspend entry (s2idle)
[  136.989539] PM: Syncing filesystems ... done.
[  136.992447] Freezing user space processes ... (elapsed 0.002 seconds) done.
[  136.994916] OOM killer disabled.
[  136.994917] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  136.995975] printk: Suspending console(s) (use no_console_suspend to debug)
[  137.098573] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 5
[  137.098584] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  137.098728] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  137.098736] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  137.098952] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  137.098955] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=5
[  137.098959] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  137.099122] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  137.112525] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  137.112707] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  137.112756] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  137.112762] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[  137.112765] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  137.112814] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[  137.129032] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[... 300 lines cut ...]
[  139.799954] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.799965] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.800159] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.800169] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.800364] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.800374] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.800569] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.800579] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.800835] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.800847] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.801078] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  139.801088] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[... cut ...]
[  139.801320] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[  141.070111] ata1.00: configured for UDMA/100
[  145.849005] OOM killer enabled.
[  145.849007] Restarting tasks ... done.
[  145.923103] PM: suspend exit

```
[s3_test_5_dmesg.log](https://github.com/thesofproject/sof/files/3215204/s3_test_5_dmesg.log)
[s3_test_6_dmesg.log](https://github.com/thesofproject/sof/files/3215205/s3_test_6_dmesg.log)
[s3_test_6_logger_error.log](https://github.com/thesofproject/sof/files/3215206/s3_test_6_logger_error.log)

ClarexZhou, 2019-05-24:
Error still occurred after merge https://github.com/thesofproject/linux/pull/977

**Environment**
Kernel: 9453da1 https://github.com/lyakh/linux/tree/ipc-timeout (with PR895) +PR977
FW:b527fd7 https://github.com/lyakh/sof/tree/ipc-timeout (with PR1354)
Topology:Same as above FW repo, file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg
kconfig:
unset CONFIG_SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE
unset CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA          13106377.135417  13106377.000000 intel/cavs/hda-dma.c:639 	hda-dmac: 5 invalid channel -1

```

```
[  111.942839] PM: suspend entry (s2idle)
[  111.942842] PM: Syncing filesystems ... done.
[  111.945863] Freezing user space processes ... 
[  115.089982] Freezing of tasks aborted after 3.144 seconds (1 tasks refusing to freeze, wq_busy=0):
[  115.089995] OOM killer enabled.
[  115.089997] Restarting tasks ... done.
[  115.125056] PM: suspend exit
[  115.857025] sof-audio-pci 0000:00:1f.3: hda: prepare stream dir 0
[  115.857056] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  116.160124] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  116.160145] sof-audio-pci 0000:00:1f.3: unknown ROM status value 00000021
[  116.160453] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000080 trace 0x00000080
[  116.160468] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  116.160476] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[  116.160482] sof-audio-pci 0000:00:1f.3: ASoC: cpu DAI prepare error: -110
[  116.160488]  Analog Playback and Capture: ASoC: backend prepare failed -110
[  116.160952] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[  116.160962] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  116.464129] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[  116.464149] sof-audio-pci 0000:00:1f.3: unknown ROM status value 0000001c
[  116.464457] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000081 trace 0x00000081
[  116.464472] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  116.464520] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  116.768134] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  116.768154] sof-audio-pci 0000:00:1f.3: unknown ROM status value 0000001c
[  116.768483] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000081 trace 0x00000081
[  116.768498] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  116.768505] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[  116.768711] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
```
[wasp_s3_test_9_dmesg.log](https://github.com/thesofproject/sof/files/3216418/wasp_s3_test_9_dmesg.log)
[wasp_s3_test_9_logger_error.log](https://github.com/thesofproject/sof/files/3216419/wasp_s3_test_9_logger_error.log)

ClarexZhou, 2019-05-24:
occurred again during S3 with playback stress test in 8th.

[s3_test_8_logger_error.log](https://github.com/thesofproject/sof/files/3216443/s3_test_8_logger_error.log)
[test_7.log](https://github.com/thesofproject/sof/files/3216444/test_7.log)
[test_8.log](https://github.com/thesofproject/sof/files/3216445/test_8.log)

lgirdwood, 2019-05-24:
@ranj063 @kv2019i @lyakh I see this in every test from @ClarexZhou 
```[  116.160468] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003```
Even when TRACE is OFF.

xiulipan, 2019-05-27:
@lgirdwood 
We have different for cavs1.8 in cnl.c

https://github.com/thesofproject/linux/blob/924347de73e488fa21772af90566821694dcce55/sound/soc/sof/intel/cnl.c#L165-L181

keyonjie, 2019-05-27:
@ranj063's Kernel PR https://github.com/thesofproject/linux/pull/977 fixed this issue for me.

lgirdwood, 2019-05-28:
@xiulipan ok great, thanks for checking. It is probably worth a comment in the generically named version to state it's for 1.5 only.

xiulipan, 2019-05-28:
@lgirdwood 
Yes, the hda.c is actually a very misleading file name. We need some comments inline to identify what these files actually for.

lgirdwood, 2019-05-28:
@xiulipan agreed, we do have differences in version between 1.5, 1.8 and 2.x so having some comments to describe which version that file/function is for would avoid any confusion going onwards. Can you create a feature request for this ?

xiulipan, 2019-05-29:
@lgirdwood  @plbossart 
Done with https://github.com/thesofproject/linux/issues/987
We need some explanation about what is the mapping.

wenqingfu, 2019-06-19:
not reproducible after the thesofproject/linux#1013 et al.

closed 2019-06-19 by wenqingfu