thesofproject/sof #1423: [BUG] KWD: failed to trigger DETECT9.0 and IPC error

keqiaozhang, 2019-05-14:
**Describe the bug**
This issue happens when doing the wov capture. The arecord  process just stop there and unable to wake up(clap hands hard). 

**To Reproduce**
arecord -Dhw:0,8 -M -N -r 16000 -f s16_le -c 2 wov.wav -vvv

**Expected behavior**
keyword is detected and keyword data is draining, PCM will start to get captured data from FW

**Impact**
failed to detect keyword and capture data

**Environment**
1) kernel: sof-dev/7271e69
2) firmware: master/b73b9a
3) tplg: sof-glk-da7219-kwd.tplg
4) platform: GLK Ubuntu18.04
5) Reproducibility Rate.100%

**Screenshots or console output**
```
test@test-Bobba:~$ arecord -Dhw:0,8 -M -N -r 16000 -f s16_le -c 2 wov.wav -vvv
Recording WAVE 'wov.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
Hardware PCM card 0 'sof-glkda7219max' device 8 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 16
  buffer_size  : 8016
  period_size  : 2016
  period_time  : 126000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2016
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8016
  silence_threshold: 0
  silence_size : 0
  boundary     : 9025213653250473984
  appl_ptr     : 0
  hw_ptr       : 0
mmap_area[0] = 0x7f65b4c6e000,0,32 (16)
mmap_area[1] = 0x7f65b4c6e000,16,32 (16)
Max peak (4032 samples): 0x00000001 #                    0%
Max peak (4032 samples): 0x00000001 #                    0%
Max peak (4032 samples): 0x00000001 #                    0%
Max peak (4032 samples): 0x00000001 #                    0%
Max peak (4032 samples): 0x00000001 #                    0%
Max peak (320 samples): 0x00000001 #                    0%
```

**dmesg**
```
[   78.860258] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   78.860742] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   78.860744] sof-audio-pci 0000:00:0e.0: pcm: stream dir 1, posn mailbox offset is 790528
[   78.872837] sof-audio-pci 0000:00:0e.0: received event 1 for widget DETECT9.0
[   78.872842] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   78.872977] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   78.872979] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   78.873107] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   78.873109] sof-audio-pci 0000:00:0e.0: received event 2 for widget DETECT9.0
[   78.874204] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 8 dir 1 cmd 1
[   78.874220] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   78.874352] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   80.188777] sof-audio-pci 0000:00:0e.0: ipc rx: 0x50050039: GLB_COMP_MSG: unknown type
[   80.188780] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x50050039: GLB_COMP_MSG: unknown type
[   90.753582] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 8 dir 1 cmd 0
[   90.753602] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   90.753745] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   90.753780] sof-audio-pci 0000:00:0e.0: pcm: free stream 8 dir 1
[   90.753782] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   90.754158] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   90.765691] sof-audio-pci 0000:00:0e.0: pcm: close stream 8 dir 1
[   90.765839] sof-audio-pci 0000:00:0e.0: received event 4 for widget DETECT9.0
[   90.765842] sof-audio-pci 0000:00:0e.0: received event 8 for widget DETECT9.0
[   90.765845] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   90.765955] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60050000 size 12
[   90.765960] sof-audio-pci 0000:00:0e.0: error: failed to trigger DETECT9.0
[   90.765961] sof-audio-pci 0000:00:0e.0: error: failed to trigger widget DETECT9.0
[   90.765964] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   90.766319] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   92.920439] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   92.920500] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
```
** trace**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          MEM             51778.281250     51778.281250      src/lib/alloc.c:271    error: 342 blocks needed for allocation but only 341 blocks are remaining
    0      1         COMP          11964983.072917  11913205.000000 rc/audio/component.c:161    comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC          11964988.593750         5.520833 src/audio/pipeline.c:606    pipeline_trigger() error: ret = -22, host->comp.id = 57, cmd = 0
    0      1          IPC          11964994.270833         5.677083    src/ipc/handler.c:513    ipc: comp 57 trigger 0x50000 failed -22
```

[dmesg.log](https://github.com/thesofproject/sof/files/3176644/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3176645/logger.log)
[logger-t.log](https://github.com/thesofproject/sof/files/3176646/logger-t.log)

tlauda, 2019-05-14:
@keqiaozhang Traces are missing in logger-t.log. File logger.log contains errors which are not seen in the logger-t.log. Could you make sure you are not using FW with additional custom changes as in the FW panic issue?

keqiaozhang, 2019-05-15:
@tlauda 
The attached trace logs are exactly the logger outputs, I just confirmed it again and got the same traces. the code base is clean without PRs or workarounds.

tlauda, 2019-05-15:
@keqiaozhang Have you tried with #1383? It looks like there is some mismatch between components' states.

keqiaozhang, 2019-05-15:
@tlauda 
Yes, this issue still can be reproduced with #1383 .

Test recipe:
kernel: sof-dev/44443c
firmware: glk-012-stable-branch/59ad3 +pr #1383

keqiaozhang, 2019-05-15:
This issue is not reproducible with PR #1434 .

wenqingfu, 2019-05-15:
1434 goes to a dev branch, how about the status of these commits go into master branch? @keyonjie

keyonjie, 2019-05-16:
> 1434 goes to a dev branch, how about the status of these commits go into master branch? @keyonjie

@wenqingfu there are PRs for all those 6 commits, but are waiting for reviewing and comments, PR#1365 is controversial, @tlauda suggest to use #1383(need update to add trigger dis-propagate, I have commented there) to replace it.

keyonjie, 2019-05-16:
sorry, @wenqingfu just realized that #1365 has been merged by Liam, thanks. So now only #1427 #1435 #1426 #1430 and #1415 are on the fly.

keqiaozhang, 2019-05-17:
This issue is fixed in glk-012-stable-branch/90700b.

closed 2019-05-17 by keqiaozhang