thesofproject/sof #1424: [BUG] Different IPC timeouts resulting in FW returning code 00000005

sathyap-chrome, 2019-05-14:
**Describe the bug**
During warm boot cycling on CML platform we are reproducing the below IPC timeout resulting in topology load failing.

**To Reproduce**

Try warm boot cycling on CML platform
Upon boot, we fail at loading topology with these 2 error signatures

sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40070000 size 12
sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005

sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x30100000 size 48
sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005

**Expected behavior**
Topology load should be consistent on all boots.

**Impact**
Audio not working randomly on some boot cycles.

**Environment**
1) Seen with upstream patches of SOF along with cml_drop 0004 branch for topology and firmware.
      Same on seen on cml_drop 0002 as well along with attached topology and SOFv0.2 kernel.
2) Name of the topology file : Topology file : sof-cml-rt5682-max98357a.tplg
      For 0002 branch, the topology m4 is attached.
3) Name of the platform(s) on which the bug is observed.
      CML platform
4) Reproducibility Rate. If you can only reproduce it randomly, it’s useful to report how many times the bug has been reproduced vs. the number of attempts it’s taken to reproduce the bug.
     Warm boot cycling of 500 cycles reproduces this issue 2 - 5 times.

Attaching 2 dmesg.txt from the 2 instances of the error.
[dmesg_240_iteration.txt](https://github.com/thesofproject/sof/files/3176722/dmesg_240_iteration.txt)
[dmesg_145.txt](https://github.com/thesofproject/sof/files/3176732/dmesg_145.txt)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

jajanusz, 2019-05-14:
> Seen with upstream patches of SOF along with cml_drop 0004 branch for topology and firmware.
Same on seen on cml_drop 0002 as well along with attached topology and SOFv0.2 kernel.

Please point at one commit id. Prefereably master because there were many patches for IPC since cml drop.

sathyap-chrome, 2019-05-14:
58de5323b9e1 topology: add demux topology for apl and cml 
     Compiled Topology & firmware based on cml-004-drop-stable

tlauda, 2019-05-14:
@sathyap-chrome Please also attach FW logs.

sathyap-chrome, 2019-05-14:
I will reqeust validation to run sof logger - it was not run during this cycle. Can get logs only EOD tomorrow as this tests are run overnight due to boards limitation.,

sathyap-chrome, 2019-05-16:
Attached are sof logger with corresponding dmesg logs too.
I've also attached 1 sof logger of an iteration where audio works fine.

[dmesg_failed_iteration_2.txt](https://github.com/thesofproject/sof/files/3185498/dmesg_failed_iteration_2.txt)
[sof_logger_failed_iteration_2.txt](https://github.com/thesofproject/sof/files/3185499/sof_logger_failed_iteration_2.txt)
[sof_logger_failed_iteration1.txt](https://github.com/thesofproject/sof/files/3185500/sof_logger_failed_iteration1.txt)
[sof_logger_workingbootcycle.txt](https://github.com/thesofproject/sof/files/3185501/sof_logger_workingbootcycle.txt)
[dmesg_failed_iteration_1.txt](https://github.com/thesofproject/sof/files/3185502/dmesg_failed_iteration_1.txt)

sathyap-chrome, 2019-05-21:
@tlauda 
Hi, gentle ping. Are the logs in previous comment useful.
is there some temp fix i can try for the same ?

wenqingfu, 2019-05-25:
one of the error report is same as thesofproject/linux#924 but on different platform. (I2S vs HDA)

mengdonglin, 2019-06-28:
@sathyap-chrome  @ranj063  Closing this issue because we cannot reproduce it in recent regression test for weeks.  All driver fixes have been merged to sof-dev. 
@sathyap-chrome Please reopen it if you can still reproduce it after fetching all driver updates.

closed 2019-06-28 by mengdonglin