thesofproject/sof #1429: [IMPROVEMENT] Add IPC structs' fields documentation

marcinx-zielinski, 2019-05-14:
We need to have our IPCs documented to know clearly what is the purpose/role of and possibly constraints on each of the IPC struct's fields. For example, now we have:
`struct sof_ipc_comp_host {
	struct sof_ipc_comp comp;
	struct sof_ipc_comp_config config;
	uint32_t direction;	/**< SOF_IPC_STREAM_ */
	uint32_t no_irq;	/**< don't send periodic IRQ to host/DSP */
	uint32_t dmac_config; /**< DMA engine specific */
} __attribute__((packed));`
This doc is somewhat cryptic - it assumes that a programmer has in-depth code knowledge. Another example:
`struct sof_ipc_pipe_ready {
	struct sof_ipc_cmd_hdr hdr;
	uint32_t comp_id;
} __attribute__((packed));`
What is the acceptable range of _comp_id_?

lgirdwood, 2020-01-27:
@lbetlej I'll move this into v1.6

lgirdwood, 2020-08-18:
This is partially done. Move to v1.7.

lgirdwood, 2021-01-12:
@mwasko @mrajwa how are we now for IPC field docs ? I've seen updates but not sure if we can close yet ?

mrajwa, 2021-01-12:
Sorry I have no clue, @mwasko do you follow this?

lgirdwood, 2021-05-10:
Now started, some structure have fields for doxygen but some are still missing. To be completed in v1.9 as part IPC abstraction.

kv2019i, 2023-02-15:
I'll go ahead and close this even if this was only partially done.

closed 2023-02-15 by kv2019i