thesofproject/sof #1437: [BUG]WoV: there will be noise occasionally in the first 2 seconds

keqiaozhang, 2019-05-15:
**Describe the bug**
During the validation, I found that there're some noise in the draining data(first 2 seconds) occasionally. There're no obvious errors in dmesg and trace. The noise wav is attached as below.
[wov.zip](https://github.com/thesofproject/sof/files/3181813/wov.zip)

**To Reproduce**
1.  play a sound on the speaker first as the capture input.
 $ aplay -Dhw:0,0 -f dat -c 2 audio.wav &
2. do dmic capture and clap your hands hard to wake up arecord.
 $ arecord -Dhw:0,8 -M -N -r 16000 -f s16_le -c 2 wov.wav -vvv -d 10
3. after capture is done, kill the aplay and check the recorded wav with paplay
$ pkill -9 aplay
$ paplay wov.wav
4. then you will hear the noise in the first 2 seconds

**Expected behavior**
The recorded sound should be clear without noise.

**Impact**
there're some noise in the first seconds.

**Environment**
1) kernel: sof-dev/44443 + https://github.com/thesofproject/linux/pull/942
2) sof: glk-012-stable-branch/59ad3c + PR #1434
3) Platform: GLK Ubuntu18.04
4) Reproducibility Rate： 10%~20%

**Screenshots or console output**
no error output

[dmesg.log](https://github.com/thesofproject/sof/files/3181856/dmesg.log)
[logger-t.log](https://github.com/thesofproject/sof/files/3181857/logger-t.log)

keyonjie, 2019-05-15:
@mrajwa looks there is noise in the draining data sometimes, can you help check?

keyonjie, 2019-05-16:
@keqiaozhang can you check if https://github.com/thesofproject/sof/pull/1444 is helpful for this issue?

mrajwa, 2019-05-16:
@keqiaozhang, I can confirm I have heard similar issue some time ago but with newest fixes from integration branch it is no longer reproducible.

keqiaozhang, 2019-05-17:
@mrajwa 
Unfortunately, this issue is still reproducible to me. I tested it with top glk-012-stable-branch, but 16 out of 30 are failures. I attached 2 of the failures for your reference.
[test-22.zip](https://github.com/thesofproject/sof/files/3190086/test-22.zip)
[test-28.zip](https://github.com/thesofproject/sof/files/3190089/test-28.zip)

Test recipe:
kernel: sof-dev/0f3e6
firmeware:glk-012-stable-branch/90700b + PR #1446

keqiaozhang, 2019-05-20:
This issue is fixed on glk-012-stable-branch/d7c4d1.  This issue can be closed.

closed 2019-05-20 by keqiaozhang