thesofproject/sof #1443: [BUG][BYT] nocodec loopback recording has only silence

fredoh9, 2019-05-16:
**Describe the bug**
For BYT Minnowboard, Loopback recording has only silence. Daily CI build, 5.1.0-daily-nocodec-20190516 is used. Loopback in APL UP2 nocodec mode worked fine. Linux driver parse the tplg and set ssp.quirks(=64).

[   13.852957] sof-audio-acpi 80860F28:01: tplg: config SSP2 fmt 0x4001 mclk 19200000 bclk 2400000 fclk 48000 width (24)25 slots 2 mclk id 0 quirks 64

sof dsp may not utilize this quirks bit.

**To Reproduce**
Try loopback recording and check output wave file.
Terminal1)
$ arecord -vv -Dhw:0,0 -f S16_LE -c 2 -r 48000 mb_nocodec_s16.wav
Terminal2)
$ aplay -vv -Dhw:0,0 -f S16_LE -c 2 -r 48000 ~/Music/short_48000.wav

**Expected behavior**
Recorded wave file should have playback sound.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
BYT Minnowboard Nocodec mode.

**Screenshots or console output**
[mb_0515_nocodec_booting_dmesg.log](https://github.com/thesofproject/sof/files/3184906/mb_0515_nocodec_booting_dmesg.log)
[mb_0515_nocodec_loopback.log](https://github.com/thesofproject/sof/files/3184907/mb_0515_nocodec_loopback.log)
[mb_0515_sof_logger.log](https://github.com/thesofproject/sof/files/3184908/mb_0515_sof_logger.log)

Jiangxinx, 2019-05-16:
I can't  observe this issue on daily test.

***recipe***
linux (topic/sof-dev) : 465c8e40
sof (master) : 73cefecd
topology : sof-byt-rt5651.tplg & sof-byt-nocodec.tplg

fredoh9, 2019-05-16:
@Jiangxinx can you share output file of lookback? I don't think DSP in BYT support LBM. This is problem on nocodec mode. I don't think dailytest test loopback( playback+recording).

Jiangxinx, 2019-05-17:
@fredoh9 Do you have reworked you device? Here are my environment and test step.
 ***Environment***
[image]

linux (topic/sof-dev) : 2291b8c2
sof (master) : f04bc83f
topology : sof-byt-nocodec.tplg

***Step***
aplay -Dhw:0,0 -fS16_LE -c2 -r 48000 audio.wav & arecord -Dhw:0,0 -c2 -fS16_LE test.wav -vv -i

***record file***

xiulipan, 2019-05-17:
@Jiangxinx @fredoh9 
All nocodec should work with nocodec tplg to make loopback in SSP drvier.
@fredoh9 
Please fix the missing part in byt/ssp.c and hsw/ssp.c to set lbm bit in ssrc1

jajanusz, 2019-06-07:
@xiulipan @Jiangxinx @fredoh9 More like you need quirks support in byt/hsw ssp.c at all. Why there are no  ssp quirks for these platforms?

xiulipan, 2019-06-10:
@jajanusz 
Thanks for the tips,I checked the code and we did missed some loopback set up for BYT/HSW.
@fredoh9 Could you check patch #1538 from @jajanusz

fredoh9, 2019-06-15:
I tested #1538 on MB NOCODEC mode. it worked fine. Looks good to me. Thank you for the fix.

PR #1538 (2019-06-15): ssp: quirks for byt and hsw
Fixes #1443 

It adds quirk support that was so far enabled only for CAVS platforms. Registers for BYT & HSW were already there.
closed 2019-06-15 by jajanusz