thesofproject/sof #1454: [BUG]EC： IPC error warning after booting on WHL-I2S

keqiaozhang, 2019-05-20:
**Describe the bug**
There's one IPC error after booting on WHL-I2S, this ipc error is for  0x60050000 size 12.
But playback and capture are still functional.

**To Reproduce**
1. reboot the system and check the dmesg and trace directly.

**Expected behavior**
No errors or warning.

**Impact**
no direct impact so far. 

**Environment**
1) Kernel: sof-dev/901cd+PR870
2) Firmware: cml-004-drop-stable/3d1a95 + PR #1438 
3) Topology：sof-cml-demux-rt5682.tplg
4) Platform: WHL-I2S mode
5) Reproducibility Rate. 100%.

**dmesg**
```
[    4.733508] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50020000: GLB_COMP_MSG: GET_VALUE
[    4.736712] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[    4.736718] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[    4.736845] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60040000 size 12
[    4.736849]  Port1: ASoC: trigger FE failed -22
[    4.736851] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[    4.736855] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[    4.736932] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[    4.736934]  Port1: ASoC: trigger FE failed -22
[    4.737135] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 1
[    4.737137] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[    4.737138] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[    4.737138] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
```
**Trace**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1         COMP         244131560.729167 244131568.000000 rc/audio/component.c:147    comp_set_state() error: wrong state = 1, COMP_TRIGGER_START
    0      1          IPC         244131567.395833         6.666667 src/audio/pipeline.c:625    pipeline_trigger() error: ret = -22, host->comp.id = 0, cmd = 1
    0      1          IPC         244131574.427083         7.031250    src/ipc/handler.c:513    ipc: comp 0 trigger 0x40000 failed -22
    0      1         COMP         244131674.270833        99.843750 rc/audio/component.c:168    comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC         244131680.468750         6.197917 src/audio/pipeline.c:625    pipeline_trigger() error: ret = -22, host->comp.id = 0, cmd = 0
    0      1          IPC         244131687.291667         6.822917    src/ipc/handler.c:513    ipc: comp 0 trigger 0x50000 failed -22
```

[dmesg.log](https://github.com/thesofproject/sof/files/3196073/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3196074/logger.log)
[logger-t.log](https://github.com/thesofproject/sof/files/3196075/logger-t.log)

keqiaozhang, 2019-05-21:
This issue is introduced by Pulseaudio,  no such error when PA is disabled.

lgirdwood, 2019-05-27:
@keqiaozhang always best to check the FW log when you have an IPC driver error for the **real** cause of the issue. Updated title.

YvonneYang2, 2019-07-22:
Issue not reproduced on whl-i2s with test recipe as below. So close the bug.

Test recipe:
Firmware: a513ac72 https://github.com/thesofproject/sof/commits/master
Kernel: 44c31df1 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology:file: sof-cml-rt5682.tplg link from sof-whl-rt5682.tplg
platform: whl-i2s

closed 2019-07-22 by YvonneYang2