thesofproject/sof #1463: [BUG] CML : S3 failed due to ipc error -110

sathyap-chrome, 2019-05-21:
**Describe the bug**
When S3 is attempted, there is IPC timeout during resume. And subsequent S3 also fails.
What have you tried to diagnose or workaround this issue?
From the signature, this seems to be similar to https://github.com/thesofproject/sof/issues/1342
Please also read https://thesofproject.github.io/latest/howtos/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Enable S3 (instead of soix)
echo 0 > /var/lib/power_manager/suspend_to_idle
restart powerd
Then run suspend_stress_test -c 1
dmesg shows the IPC errors

**Expected behavior**
DSP should resume without issues

**Impact**
System stability as S3 criteria is broken

**Environment**
1) cml_drop004 FW and Topology and v6 kernel patches taken directly from UPSTREAM
2) sof-cml-rt5682-max98357a.tplg
3) CML.
4) Reproducibility Rate. : Always
**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

[s3_failure_sof_print.txt](https://github.com/thesofproject/sof/files/3203191/s3_failure_sof_print.txt)

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

michalgrodzicki, 2019-05-21:
Hi @sathyap-chrome, 
could you share FW/kernel git hashes which you used here?
Are you using typical HW configuration with I2S mode?

sathyap-chrome, 2019-05-21:
The configuration is the typical i2s mode.
Kernel is latest set of upstream patches which is v6 patches of sof.

FW and Topology : (remotes/origin/cml-004-drop-stable) 3d1a95d414ab topology: add demux topology for apl and cml

keqiaozhang, 2019-05-27:
@sathyap-chrome 
Could you please share your kernel branch(Repo URL). I will try to reproduce this issue on my test device.

lgirdwood, 2019-05-27:
@mengdonglin @wenqingfu fyi CTX_SAVE IPC timeout.

sathyap-chrome, 2019-05-29:
@keqiaozhang we had created a branch by pulling the upstream sof patches. but we are trying to align to sof-dev branch and reproduce the issue. 

1 important observation we made yesterday: if we enable no_console_suspend and earlyprintk in bootargs, the issue is not happening. In case you have them enabled, please remove and try.

ranj063, 2019-05-29:
> 1 important observation we made yesterday: if we enable no_console_suspend and earlyprintk in bootargs, the issue is not happening. In case you have them enabled, please remove and try.

@sathyap-chrome is this just an experiment or is this how this is supposed to be tested?

sathyap-chrome, 2019-05-29:
@ranj063  by default chrome boot args dont use the no console suspend and early printk. yesterday for other issue, we tried that and saw the S3 working perfectly fine. then we removed the above bootargs and we hit this issue right away. just wanted to share observation if incase in any of your testing you have these bootargs, we should remove and test.

lgirdwood, 2019-05-29:
@sathyap-chrome both those args will impact timing and no_console_suspend will probably impact the overall system PM state. Is there any impact if you remove the early printk (and keep console) ? and any impact the other way around ? This may shed light on whether it's more timing or PM....

sathyap-chrome, 2019-05-29:
@lgirdwood  Agreed. we had limited EVTs to try the test multiple times. I can give this data tomorrow.
Also i tried with the PR#943 kernel patch on my tip and tried very few cycles of S3. I didnt observe ipc timeouts but there were some other issues ( still not triaged if its audio or not) . will keep you posted by EOD tomorrow

kv2019i, 2019-05-31:
Issue very likely to be problem with MSI interrupt mode and fixed by https://github.com/keyonjie/linux/commit/08be9af66613b1298c702ebb15d6bc4d15b77cc9 So not a SOF/FW bug.

ranj063, 2019-05-31:
@kv2019i I checked with @sathyap-chrome and he mentioned that the issue was fixed with https://github.com/thesofproject/linux/pull/943

@sathyap-chrome could you please confirm and close the issue if that is still correct?

sathyap-chrome, 2019-05-31:
I would like to run more stability before closing this issue.
Will keep you posted.

@ranj063 is it good idea to still take keyonjie/linux@08be9af ?
Even if i dont hit issues after picking PR#943 ?

ranj063, 2019-05-31:
@sathyap-chrome we're still discussing if disabling MSI for all platforms is the way to go. If you see issues with CTX_SAVE, please try it and let us know if it helps. Otherwise, I'd say lets wait till it is merged to sof-dev.

mengdonglin, 2019-06-28:
@sathyap-chrome  @ranj063  Closing this issue because we cannot reproduce it in recent regression test for weeks.  All driver fixes have been merged to sof-dev. 
@sathyap-chrome Please reopen it if you can still reproduce it after fetching all driver updates.

closed 2019-06-28 by mengdonglin