thesofproject/sof #1467: [BUG] KD sporadic glitches

wwittbrx, 2019-05-22:
### Description

Stress tests of KD with DMIC test case reveal sporadic glitches with reproduction rate of ~0.2%. Logs and waves of failed iterations attached.

### Recipe

**FW**
https://github.com/thesofproject/sof/tree/glk-012-stable-branch
https://github.com/thesofproject/sof/commit/ddde684d7296f3059e195d4dfe9c832367fcddfb

**Platform**
APL RVP

### Reproduction steps

1. Run test_03_kd_with_dmic.py with iterations set to 1000 (may need more iterations for reproduction depending on luck)

### Logs and files

[logs.zip](https://github.com/thesofproject/sof/files/3207463/logs.zip)

lgirdwood, 2019-05-22:
@wwittbrx best to always paste the error message from the logs in the description (only a few lines needed), that ways it obvious whether its a pipeline, dmic, dma bug etc. The issue subject can also be renamed to reflect the error in the logs.

wwittbrx, 2019-05-23:
Unfortunately the logs don't contain any errors so I couldn't do that. I have added wave screenshots for visibility without going into logs zip. I am also currently stress testing simple DMIC capture to check if this is indeed KD related or just DMIC related. Will update as soon as I am confident with findings.

wwittbrx, 2019-05-23:
After over a 1000 iterations of simple DMIC capture and not detecting any glitches I would assume that the problem is with KD itself.

mrajwa, 2019-05-23:
@wwittbrx, just to clarify, glitches occur not only in first 2100 ms but also later?

mrajwa, 2019-05-23:
also, can you provide information about history_depth and preamble you used for this test?

mrajwa, 2019-05-24:
I haven't debug this issue yet because we have higher priority issue (the occurrence of this one is very low + it wasn't noticed outside slim driver). However at first glance it suggest me a **dmic** issue, especially when I look at the first screenshot, the waveform is cut but then restores with **damped** amplitude. After another while it is normal again. @singalsu what do you think about this?

singalsu, 2019-05-24:
@mrajwa I'll take a look into the wavs, thanks for attaching those as well.

singalsu, 2019-05-24:
The only thing looking like this a buggy driver could do would be to launch in mid-recording the gain ramp (shouldn't happen). Adding a trace print to function dmic_work() would show in trace when the DMIC HW gain registers are updated.

Though gain ramp controls left and right channels simultaneously and in this code the channels are updated next to each other so something higher priority only cause the code to pause.

It this test acoustical or done with DMIC injector? Could the injector cause issues (is it slave to the DMIC interface like a real microphone is)?

singalsu, 2019-05-24:
Based on log file seems that mic clock is set to 800 kHz. If the test was acoustical with real microphone it would be good to check the datasheet of the microphone that (a low-power mode clock) 800 kHz is within normal operation range.

wwittbrx, 2019-05-27:
Test is performed with DMIC injector. As I said in the earlier comment, I don't think the problem is with DMIC itself, as I stress tested DMIC and glitches do not happen without KD pipeline. Preamble time is set to 500, and history depth is set to 0, so I guess it takes the default value.

lgirdwood, 2019-05-28:
@wwittbrx do we have a meaningful reproduction rate or are there methods to increase the reproduction rate ? It could be KD logic related since the sine waveforms do seem to join up nicely from your screenshot. i.e. we stop processing for several periods then restart.

wwittbrx, 2019-05-29:
Reproduction rate is ~0.2%. I don't know of any methods to increase it. We could maybe reduce individual iteration execution time to run stress tests faster, I will provide more info if successful.

mrajwa, 2019-06-03:
> Preamble time is set to 500, and history depth is set to 0

@wwittbrx, it means detection is active after 500ms and you will drain **zero** bytes from history buffer, is that what you really want in this test? Also the default `history depth` shall be set to 2100 ms.

wwittbrx, 2019-06-03:
@mrajwa I meant that I don't set history depth at all in this test. That means that in the IPC we send zeroes for this field. I was under the impression that if we do that, we will fall back to default defined in FW code, which as you said is 2100.

mrajwa, 2019-07-23:
Still reproducible?

wwittbrx, 2019-10-07:
This issue cannot be reproduced with d1f4320, closing.

closed 2019-10-07 by wwittbrx