thesofproject/sof #1471: [FEATURE] Refactor exported headers + add folder for tools.

jajanusz, 2019-05-24:
**Is your feature request related to a problem? Please describe.**
Our current headers folders structure seems to be bad in some scenarios, it recently showed up f.e. in these PRs:
#1405 - we ended up with workaround, but not everything going to be that simple and we need some kind of place for logic shared between FW and user-space apps (but not by kernel).
#1351 - there are definitions that are needed by FW and by tools but not by kernel, we can add everything to be in uapi, but this way we going to have a lot of definitions on kernel side that are not used by kernel at all, because they are just for tools that generate binary blobs.

Also should `uapi` folder be named like that? - In kernel it is for user-space. Here this name is misleading because it's for kernel/driver and what really end-up in user-space may be headers for tools - but still it's not the same, because they don't use it in the same way as user-space apps use kernel's uapi.

**Describe the solution you'd like**
I'd like to have more fine-grained headers classification with clear purpose.

lgirdwood, 2019-05-27:
@jajanusz please describe what you propose.

jajanusz, 2019-05-27:
@plbossart already propsed something there: https://github.com/thesofproject/sof/pull/1405#discussion_r282882355 ->
>@lgirdwood any objection to create a 3rd directory, we'd have
ipc
user (can be renamed as kernel if needed)
apps

Personally I'd make it saying more info about what's in it, instead of for whom it is, f.e.:
```
include
├── api (it's just api, we don't have "userspace" api)
│   ├── ipc (obvious, ipc with our FW)
│   └── meta (info about FW internals, manifest, modules, or some other name, depends what is there)
│       ├── kernel (or user/driver, indicating it's FW's metadata necessary for driver)
│       └── apps (or non-kernel/others/etc, data not needed by driver, used by tools/sdk)
└── sof (internal, project-wide headers, used across FW modules)
```
But I wrote most of my code as guy from userspace, so I may look at it from bad point. 
Also I'm not sure how many of these things are really needed on driver side, so maybe it may be split differently, but you should get the idea.

lgirdwood, 2019-05-28:
@jajanusz Don't need the api/meta subdirs.
```
include
│ ── ipc (obvious, ipc with our FW)
│ ── kernel (or user/driver, indicating it's FW's metadata necessary for driver )
│ ── user (or non-kernel/others/etc, data not needed by driver, used by tools/sdk)
└── sof (internal, project-wide headers, used across FW modules)
```

jajanusz, 2019-05-28:
@lgirdwood So you want to make it more flat ? (There is uapi folder missing)

keyonjie, 2019-05-28:
> @jajanusz Don't need the api/meta subdirs.
> 
> ```
> include
> │ ── ipc (obvious, ipc with our FW)
> │ ── kernel (or user/driver, indicating it's FW's metadata necessary for driver )

do we have any non-ipc interfaces needed for kernel? otherwise, only "ipc" and "meta" is good?

> │ ── user (or non-kernel/others/etc, data not needed by driver, used by tools/sdk)
> └── sof (internal, project-wide headers, used across FW modules)
> ```

jajanusz, 2019-05-28:
> do we have any non-ipc interfaces needed for kernel? otherwise, only "ipc" and "meta" is good?

I don't know if they are needed by kernel, but they are copied to kernel repo - f.e. manifest metadata, some tokens.

lgirdwood, 2019-05-28:
@jajanusz yes since you are restructuring. But the make install rule must prefix all these directories with "sof/" so that there is no collisions with other headers from other SW.

jajanusz, 2019-06-04:
done in #1504

closed 2019-06-04 by jajanusz