thesofproject/sof #1475: [BUG] 4 channel DMIC recording does not work

wwittbrx, 2019-05-27:
### Description

4 channel DMIC recording does not work on GLK KD release branch. First two channels look alright, others are silent or incorrect. Neither sof-logger or dmesg show any errors as far as I can tell. Wave file and logs attached in zip archive.

This also seems to be the case on master branch of the firmware.
https://github.com/thesofproject/sof/commit/a26c51a43e436a8da15fdad39d61e13bd4cdf12f

[image]

> Recording WAVE 'record_4_channels.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Channels 4
Hardware PCM card 0 'sof-glkda7219max' device 99 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 4
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 8184
  period_size  : 2040
  period_time  : 42500
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2040
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8184
  silence_threshold: 0
  silence_size : 0
  boundary     : 9214364837600034816
  appl_ptr     : 0
  hw_ptr       : 0

### Recipe

**ChromeOS**
76.0.3781.0 (Platform 12191.0.0)

**Kernel**
https://github.com/thesofproject/linux/commit/18a233f3f676a98dde00947535d99ab1a54da340

**FW and TPLG**
https://github.com/thesofproject/sof/commit/ddde684d7296f3059e195d4dfe9c832367fcddfb
sof-glk-da-7219-kwd.tplg

### Reproduction steps

1.  arecord -D hw:0,99 -f dat -c 4 record_4_channels.wav -d 10 -vv
2. Check recorded wave with Audacity/similar

### Logs and files

[logs.zip](https://github.com/thesofproject/sof/files/3222402/logs.zip)

singalsu, 2019-05-27:
@wwittbrx How many microphones does this device have?

The channel 3 is strange, it sounds as just noise. If the device would be with two microphones I'd expect it to be like channel 4. Channel 4 looks like normal capture from missing microphone.

Even if Mic3 would be broken (if 3 microphones in the device) I'd expect channel 4 to get copy of the same data.

Can you provide here or directly by email more information about this device.

naveen-manohar, 2019-06-10:
@wwittbrx This is similar to the observations filed in https://github.com/thesofproject/sof/issues/686
And its not an issue if channel-3 & 4 has negative DC or null on a board with only 2 dmics connected(as per board schematics).

@singalsu: From the board schematics, i dont observe DMIC-3 and DMIC-4 are connected, may be .brd file can explain more on why we observe null for ch-3 & -ve DC for ch-4

singalsu, 2019-06-14:
@naveen-manohar Are devices of this model producing similar capture waveform, especially for channel 3 (some fairly silent noise, not null) that's quite different from channel 4 (large dc value, as expected for not connected microphone)?

singalsu, 2019-10-09:
@wwittbrx This is labeled as P1 bug but with no further input it's hard understand what happens. Can we close this if it was just one possibly broken device?

tlauda, 2019-10-25:
@wwittbrx To be closed?

lgirdwood, 2019-11-04:
@wwittbrx ping

closed 2019-11-05 by wwittbrx