thesofproject/sof #1485: [BUG] [WHL] SOF Oops / panic when performing an aplay + S3 stress test

lyakh, 2019-05-28:
Below is a decoded exception dump, which I've got multiple times on a Mantis and a Wasp laptops, running ipc-timeout branches:
```
#0  0xbe057400 in literals ()
#1  0xbe0534b4 in memcpy (n=1480, src=0x28, dest=<optimized out>) at .../sof/src/lib/lib.c:66
#2  arch_memcpy_s (src_size=1480, src=0x28, dest_size=393253, dest=<optimized out>) at .../sof/src/arch/xtensa/include/arch/string.h:80
#3  memcpy_s (dest=<optimized out>, dest_size=393253, src=0x28, src_size=1480) at .../sof/src/lib/lib.c:87
#4  0xbe0559ae in fill_core_dump (epc1=0xbe2eca40, stack_ptr=<optimized out>, ps=393253, oops=0xbe008800) at .../sof/src/arch/xtensa/include/arch/sof.h:88
#5  arch_dump_regs (epc1=0xbe2eca40, stack_ptr=<optimized out>, ps=393253) at .../sof/src/arch/xtensa/include/arch/sof.h:96
#6  panic_rewind (p=<optimized out>, stack_rewind_frames=<optimized out>, panic_info=<optimized out>, data=0xbe2eca40) at .../sof/src/lib/panic.c:85
#7  0xbe041910 in exception () at .../sof/src/arch/xtensa/include/arch/init.h:55
#8  0xbe056aa4 in _GeneralException ()
#9  0xbe04c296 in io_reg_read (reg=3045142744) at .../sof/src/include/sof/io.h:51
#10 host_dma_reg_read (reg=0, chan=4261724822, dma=0x180) at .../sof/src/drivers/intel/cavs/hda-dma.c:182
#11 hda_dma_start (dma=0x180, channel=-33242474) at .../sof/src/drivers/intel/cavs/hda-dma.c:560
#12 0xbe049824 in dma_start (channel=<optimized out>, dma=<optimized out>) at .../sof/src/include/sof/dma.h:264
#13 dai_copy (dev=0xbe0b1500) at .../sof/src/audio/dai.c:599
#14 0xbe04251e in comp_copy (dev=0xbe0b1500) at .../sof/src/include/sof/audio/component.h:480
#15 pipeline_comp_copy (current=0xbe0b1500, data=<optimized out>, dir=1) at .../sof/src/audio/pipeline.c:683
#16 0xbe042c1d in pipeline_copy (p=0xbe0b1600) at .../sof/src/audio/pipeline.c:728
#17 pipeline_task (arg=0xbe0b1600) at .../sof/src/audio/pipeline.c:915
#18 0xbe053fc5 in run_ll (flags=<synthetic pointer>, queue=0xbe08ac40) at .../sof/src/lib/ll_schedule.c:218
#19 queue_run (data=0xbe08ac40) at .../sof/src/lib/ll_schedule.c:331
#20 0xbe04cae0 in irq_lvl2_handler (ilxmsd=0, ilxmcd=<optimized out>, ilxsd=<optimized out>, level=6, data=0xbe05ce00 <dsp_irq>) at .../sof/src/drivers/intel/cavs/interrupt.c:103
#21 irq_lvl2_level2_handler (data=0xbe05ce00 <dsp_irq>) at .../sof/src/drivers/intel/cavs/interrupt.c:126
#22 0xbe05715d in _Level2FromVector ()
#23 0xbe05690b in arch_wait_for_interrupt (level=0) at .../sof/src/arch/xtensa/include/arch/wait.h:73
#24 wait_for_interrupt (level=0) at .../sof/src/include/sof/wait.h:58
#25 do_task_master_core (sof=0x5) at .../sof/src/tasks/audio.c:84
#26 0xbe050e4a in master_core_init (sof=0xbe066144 <sof>) at .../sof/src/init/init.c:86
#27 0xbe050e7a in main (argc=<optimized out>, argv=0xfffffffe) at .../sof/src/init/init.c:102
```

lgirdwood, 2019-05-29:
@dabekjakub one for your bug scrub

lyakh, 2019-06-17:
@tlauda shall we close this now that #1508 has been merged?

jajanusz, 2019-06-17:
@lyakh Do you still get panic? In PR you wrote that it doesn't solve issue fully.

closed 2019-06-27 by tlauda