thesofproject/sof #1497: [BUG] Coefficients files need license info

jajanusz, 2019-05-30:
Files below are missing licenses. Please add BSD license + copyright. Actually these files without that shouldn't be even in repo. Blocker for #1458 

Sorry for troubling you, but author should apply that initial license. I don't know if I can touch these files without your permission.

List of files:
```
./sof/src/include/sof/audio/coefficients/pdm_decim/pdm_decim_int32_03_4375_5100_010_095.h
./sof/src/include/sof/audio/coefficients/pdm_decim/pdm_decim_int32_03_3850_5100_010_095.h
./sof/src/include/sof/audio/coefficients/pdm_decim/pdm_decim_int32_02_4288_5100_010_095.h
./sof/src/include/sof/audio/coefficients/pdm_decim/pdm_decim_int32_06_4156_5100_010_095.h
./sof/src/include/sof/audio/coefficients/pdm_decim/pdm_decim_int32_02_4375_5100_010_095.h
./sof/src/include/sof/audio/coefficients/pdm_decim/pdm_decim_int32_08_4156_5380_010_090.h
./sof/src/include/sof/audio/coefficients/pdm_decim/pdm_decim_int32_04_4375_5100_010_095.h
./sof/src/include/sof/audio/coefficients/pdm_decim/pdm_decim_int32_05_4331_5100_010_095.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_3_1_2268_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_5_4_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_8_7_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_4_5_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_20_21_4167_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_10_21_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_32_21_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_4_3_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_3_1_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_21_20_1667_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_7_8_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_3_4_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_40_21_3968_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_21_80_3968_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_3_2_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_20_7_2976_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_6_5_4354_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_21_40_3968_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_5_6_4354_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_table.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_2_1_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_3_2_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_2_1_2268_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_1_2_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_1_3_2268_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_table.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_define.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_1_2_2268_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_8_7_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_1_3_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_25_24_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_8_7_2468_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_24_25_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_6_1_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_2_3_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_3_1_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_5_7_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_7_8_4535_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_20_21_1667_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_1_6_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_21_20_4167_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_define.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_2_3_1814_5000.h
./sof/src/include/sof/audio/coefficients/src/src_std_int32_8_21_3239_5000.h
./sof/src/include/sof/audio/coefficients/src/src_tiny_int16_1_3_1814_5000.h

```

singalsu, 2019-05-31:
@jajanusz They are automatically generated, I'll add the BSD license creation to the generator script and re-run.

jajanusz, 2019-05-31:
@singalsu I'm going to change all files to use SPDX identifiers anyway, so you can add it to your script and already use spdx instead of full license. You just need something like this on top of file:
For .c files:
```
// SPDX-License-Identifier: BSD-3-Clause
//
// Copyright(c) 2019 Intel Corporation. All rights reserved.
//
// Author: Seppo Ingalsuo <seppo.ingalsuo@linux.intel.com>
```

For .h files:
```
/* SPDX-License-Identifier: BSD-3-Clause
 *
 * Copyright(c) 2019 Intel Corporation. All rights reserved.
 *
 * Author: Seppo Ingalsuo <seppo.ingalsuo@linux.intel.com>
 */
```

lgirdwood, 2019-05-31:
@singalsu this is now a v1.3 blocker.

closed 2019-06-02 by lgirdwood

singalsu, 2019-06-03:
@jajanusz @lgirdwood Thanks for doing the update! Sorry, I didn't know it was that urgent.

I'll update the script to use the new short header.

PR #1507 (2019-06-02): license: intial spdx
As discussed in #1458 

It covers all our FW sources (c/h/S) - 3rd party files (f.e. from Tensilica) are not changed.
It add identifiers to Perl/Python/Shell scripts and Kconfig files.

I don't know if m4 and Matlab should have them. Should I add them? It's not a big deal, I'm just not sure if we have copyrights to all of them.

I used C-style comments in .S files, however they are mixed in kernel repo, so look like there is no clear convention.

This PR changes license of rimage from GPL to BSD, so @dbaluta needs to agree with that change as he is also author in files tha [... cut]