thesofproject/sof #1498: [BUG] Logger files need license info

jajanusz, 2019-05-30:
Files below are missing licenses. Please add BSD license + copyright. Actually these files without that shouldn't be even in repo. Blocker for #1458 

Sorry for troubling you, but author should apply that initial license. I don't know if I can touch these files without your permission.

List of files:
```
./sof/tools/logger/convert.c
./sof/tools/logger/logger.c
./sof/tools/logger/convert.h

```

jajanusz, 2019-05-31:
@lgirdwood @plbossart  We have GPL license in logger, should it be like this?
The same goes for rimage, should it be gpl?

lgirdwood, 2019-05-31:
@jajanusz there is no reason why it needs to be GPL. All contributions to logger are from Intel so please update to BSD.

akloniex, 2019-05-31:
License has been updated. PR was merged.
@jajanusz please close this issue.

jajanusz, 2019-05-31:
@akloniex thanks
Added in #1505

closed 2019-05-31 by jajanusz