thesofproject/sof #1502: [BUG] byt media pipe doesn't work with "192 frames per 4000us deadline"

juimonen, 2019-05-31:
**Describe the bug**
Byt media pipe plays audio with funny 4-8 times speed if you set up 
"192 frames per 4000us deadline" in topology. However,  "48 frames per 1000us deadline" works.

**To Reproduce**
Just play audio to media pipe hw:0,1

**Reproduction Rate**
Always

**Expected behavior**
Normal speed

**Impact**
Can't change scheduling params

**Environment**
1) kernel: latest, fw:latest
2) topology: sof-byt-rt5682.tplg
3) platform: byt
4) Reproducibility Rate. Always

tlauda, 2019-05-31:
@xiulipan @juimonen For EDF scheduler deadline is just maximum allowed time for the task to be executed. Media pipeline is one of couple pipes connected to the mixer, which is further connected to the DAI. DAI is set to 1 ms period, so the pipeline_copy will be executed every 1 ms and propagated to the every connected pipeline. Since media pipe is set to 192 frames we will get 4 times speed (192 frames copied every 1 ms). This is really the limitation we currently have with the scheduling of multiple connected pipelines.

xiulipan, 2019-05-31:
@tlauda 
Thansk for the quick answer!
@juimonen 
I think this issue is very worth for the root cause.

Maybe we can refine this issue to track this limitation mentioned by @tlauda

PR #1936 (2019-11-19): pipeline: schedule pipelines separately
This patch enables separate scheduling of connected
pipelines. It allows for different scheduling periods
between connected pipelines and flexible controlling
of such topologies.

Fixes #1502.
closed 2019-11-19 by lgirdwood