thesofproject/sof #1506: [BUG][GLK] HDMI failed to get afg sub nodes, no soundcard

fredoh9, 2019-05-31:
Daily build of 0530 has no soundcard due to HDMI failures.
linux-80a2cb89
sof-92fc7067

I tried daily build of 0530 onto my GLK Ubuntu RT7219 platform but I see hdmi codec failure, no soundcard. (full log attached)

[    5.018689] HDMI HDA Codec ehdaudio0D2: HDMI: failed to get afg sub nodes
[    5.018693] HDMI HDA Codec ehdaudio0D2: Failed in parse and map nid with err: -22
[    5.018710] HDMI HDA Codec: probe of ehdaudio0D2 failed with error -22

[glk_ubuntu_0530_hdmi_failure.log](https://github.com/thesofproject/sof/files/3242797/glk_ubuntu_0530_hdmi_failure.log)

fredoh9, 2019-05-31:
@ranj063 reported she found same issue in CML chromebook. And Chrome team found suspicious commit.

I tried to revert the commit id, but looks same to me.
333ee2257de8 ASoC: SOF: Intel: hda: fix the hda init chip

I can try git bisect and update this candidate commit here.

lgirdwood, 2019-05-31:
@fredoh9 this is a driver issue. Please log under linux repo.

closed 2019-05-31 by lgirdwood